sui77/rc-switch #12: Where's the wiki?

roccomuso, 2015-11-26:
How do we know which remotes are supported?

Brennenstuhl RCS 1000 N are supported?

roccomuso, 2015-12-24:
Confirmed. Brennenstuhl RCS 1000 N and Arendo outplugsboth works fine.

fingolfin, 2016-01-05:
I used the Google Code export feature to migrate the Wiki. You can find my version of it (with various cleanups already applied) here: 
  https://github.com/fingolfin/rc-switch/wiki

Since GitHub stores the Wiki as a another git repository (see https://github.com/fingolfin/rc-switch.wiki.git), it should be easy for @sui77 to add his wiki back to this repository, if he wants to.

fingolfin, 2016-01-05:
Oh, and this also migrated the issues. Just sayin'

sui77, 2016-01-07:
If only i knew how to do that. I'm still not very used to git... I've added you as Collaborator, could you do that?

fingolfin, 2016-01-07:
The first step for this requires admin access, so I can't do it, but it's also very easy: just go to the "Settings" page for this repo (i.e. https://github.com/sui77/rc-switch/settings) and enable the Wiki.
This will make a wiki link appear. Click on it, then follow the instructions to create the first page.

Once that is done, I can continue, but you can also do it yourself, it's not hard (once you know how, at least ;-): Basically, just clone your wiki https://github.com/sui/rc-switch.wiki.git  then copy the data from my version of the wiki (cloned from https://github.com/fingolfin/rc-switch.wiki.git). One can also be more fancy and also copy the history of changes I made, but that's not so important (though I'll do it, if you want, it'll only take me a few seconds)

sui77, 2016-01-07:
enabled but didn't even managed to check it out. feel free to copy/merge/import(?) yours.
danke :)

fingolfin, 2016-01-07:
Seems you missed this part of the instructions (and/or did not complete them): "Click on it, then follow the instructions to create the first page." -- this creates the Wiki repository.

Anyway, I now did that, and also transferred the Wiki. Note that I already edited it a little bit compared to what is on Google Code -- in particular, I copied all pictures, instead of relying on links to external sites (which might vanish...)

closed 2016-01-07 by fingolfin