sui77/rc-switch #13: Never reach mySwitch.available()

Kalu, 2015-12-09:
I don't know whats going wrong, it's an Arduino Uno, Pin #2. But nothing over the Serial interface. I "debugged" that it never comes into the „mySwitch.available()“ Part. I use Arduino IDE 1.6.5
If i turn Warnings in the Arduino IDE on i get several Warnings: 

Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:33:42: warning: converting to non-pointer type 'long unsigned int' from NULL [-Wconversion-null]
 unsigned long RCSwitch::nReceivedValue = NULL;
                                          ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp: In constructor 'RCSwitch::RCSwitch()':
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:49:28: warning: converting to non-pointer type 'long unsigned int' from NULL [-Wconversion-null]
   RCSwitch::nReceivedValue = NULL;
                            ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp: In member function 'void RCSwitch::switchOn(char*, int)':
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:189:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
   char* code[6] = { "00000", "10000", "01000", "00100", "00010", "00001" };
                                                                          ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:189:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:189:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:189:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:189:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:189:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp: In member function 'void RCSwitch::switchOff(char*, int)':
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:201:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
   char* code[6] = { "00000", "10000", "01000", "00100", "00010", "00001" };
                                                                          ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:201:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:201:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:201:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:201:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:201:74: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp: In member function 'char* RCSwitch::getCodeWordB(int, int, boolean)':
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:245:61: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
    char* code[5] = { "FFFF", "0FFF", "F0FF", "FF0F", "FFF0" };
                                                             ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:245:61: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:245:61: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:245:61: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:245:61: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp: In member function 'void RCSwitch::enableReceive()':
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:597:30: warning: converting to non-pointer type 'long unsigned int' from NULL [-Wconversion-null]
     RCSwitch::nReceivedValue = NULL;
                              ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:598:34: warning: converting to non-pointer type 'unsigned int' from NULL [-Wconversion-null]
     RCSwitch::nReceivedBitlength = NULL;
                                  ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp: In member function 'bool RCSwitch::available()':
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:612:38: warning: NULL used in arithmetic [-Wpointer-arith]
   return RCSwitch::nReceivedValue != NULL;
                                      ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp: In member function 'void RCSwitch::resetAvailable()':
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:616:28: warning: converting to non-pointer type 'long unsigned int' from NULL [-Wconversion-null]
   RCSwitch::nReceivedValue = NULL;
                            ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp: In static member function 'static bool RCSwitch::receiveProtocol1(unsigned int)':
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:648:25: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       for (int i = 1; i<changeCount ; i=i+2) {
                         ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp: In static member function 'static bool RCSwitch::receiveProtocol2(unsigned int)':
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:684:25: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       for (int i = 1; i<changeCount ; i=i+2) {
                         ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp: In static member function 'static bool RCSwitch::receiveProtocol3(unsigned int)':
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:722:25: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       for (int i = 1; i<changeCount ; i=i+2) {
                         ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp: In static member function 'static bool RCSwitch::receiveProtocol1(unsigned int)':
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:676:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp: In static member function 'static bool RCSwitch::receiveProtocol2(unsigned int)':
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:711:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp: In static member function 'static bool RCSwitch::receiveProtocol3(unsigned int)':
Arduino/libraries/rc-switch-2.52/RCSwitch.cpp:754:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^

rotv, 2015-12-10:
Could it be the same as issue #14?

Kalu, 2016-01-03:
I think it's kind a my mistake, i have an intertechno Set with learning Codes, and it seems the remote uses different code style. But the outlets can learn old Codes.

closed 2016-01-07 by fingolfin