sui77/rc-switch #19: Arduino Due PIN

josb86, 2016-01-07:
Hi I am using a Due but I don´t know if this library is compatible with this and if is compatible what pins use?.

roccomuso, 2016-01-08:
It should work on Arduino Due, why not. And I think you can choose whatever pin you want.

For instance:

    void setup() {
      Serial.begin(9600);
    
      // Transmitter is connected to Arduino Pin #10  
      mySwitch.enableTransmit(10);
      // Receiver connected on inerrupt 0 => that is pin #2
      mySwitch.enableReceive(0);
      
    }

sui77, 2016-01-08:
Transmission should work with any pin, receive needs a pin which is capable to process interrupts. And according to the docs the Due can handle interrupts on any digital pin. So yeah, should work.

https://www.arduino.cc/en/Reference/AttachInterrupt

josb86, 2016-01-12:
thanks it works

closed 2016-01-12 by josb86

AliTavak, 2020-08-24:
Dear firends,
I tried rc-switch reciver on DUE and get nothing! Everything is fine with uno.
I also get a warning:
"WARNING: library rc-switch claims to run on avr, esp8266 architecture(s) and may be incompatible with your current board which runs on sam architecture(s)."