sui77/rc-switch #20: Implement a learning mode

fingolfin, 2016-01-08:
I think it wouldn't be hard to implement a "learning" mode, where rc-switch detects signals it does not know before, and then replays them later.

One part of this is a modified receive function, which simply records an input, and a "replay" function which retransmits the prior signal.

This could also be helpful while debugging new protocols: Record a signal, then print it to serial.

Then, the next step is some code which takes a recorded signal, and performs a simple statistcal analysis, to determine the base pulse length, and the high/low patterns for sync, zero, one -- of course that would only work for protocols similar to the existing ones, but I believe that would cover a lot of the systems out there.

sui77, 2016-01-08:
Sounds cool. From my experience it was tricky to detect unknown signal patterns as there is a lot of noise and hard to tell where a signal starts/ends plus arduino has pretty less memory to analyse long samples.

achamely, 2016-03-23:
FWIW the example ReceiveDemo_simple is what i used to debug/clone one of my transmitters. 
The only piece missing was the pulse length. I had to add 
`      Serial.print("Delay: ");
      Serial.println( mySwitch.getReceivedDelay() );`
to the end of the current debug output (line 31) in order to get that but then I was able to successfully duplicate/clone the incoming rf codes with the mySwitch.send function
`
#include <RCSwitch.h>
RCSwitch mySwitch = RCSwitch();
void setup() {
  // Transmitter is connected to Arduino Pin #10  
  mySwitch.enableTransmit(10);
  // Optional set pulse length.
   mySwitch.setPulseLength(192);
}

void loop() {
   mySwitch.send(<received int value>, 24);
   delay(1000)
}
`

bau-sec, 2016-05-05:
I'll voice my support for the first half of this feature.  I had to modify rc-switch to do the first part (printing a recorded protocol's timings) while adding a protocol so having this available by default would have been useful.  If there's interest I could post a portion of that code here, though it's probably pretty obvious how that would be accomplished.

As for the play back a learned code, here's a guy who implemented that with a RF ceiling fan remote: http://arduinobasics.blogspot.com.au/2014/07/433-mhz-rf-module-with-arduino-tutorial_30.html . There's some weird stuff going on like hooking the receiver up to an analog input (I have the same receiver and it's digital...) but the idea is to skip the statistical analysis, just store the raw timing values.  I'm not saying that's necessarily the way to go, but here's someone who's done something similar to the second part of what your suggesting.