sui77/rc-switch #22: Problem with SCT 2260 chip

DaniLazar, 2016-01-14:
Hello,

I have bought a set of remote power sockets from ebay
http://www.ebay.com/itm/311482637595

I succeeded to read the code (decimal protocol 1 24 bit) with the  received demo of the remote controller but when trying to send it with the Arduino it dose not work!

I opened the remote and noticed that the chip is SCT 2260 that is not listed here.
Is this the problem I have?
How can I solve it?

Thanks!

fingolfin, 2016-01-17:
When you that sending does not work, what did you try exactly? Perhaps you can paste the code here?

Also, when you say you succeeded in reading the code, well, what exactly did you read? THis, too, would be useful to know. Did it match any of the codes described in the Wiki?

DaniLazar, 2016-01-17:
Well, I just used the send demo as is, just inserted in the right place in the example the decimal codes I recorded using the ReciverDemo_Simple code come with the library when pressing On/Off buttons on the remote controller.  

What do you mean by matching the codes in Wiki?
Thanks!

fingolfin, 2016-01-19:
I meant the alternative `send` methods as e.g. described on https://github.com/sui77/rc-switch/wiki/HowTo_OperateLowCostOutlets (although that list might be incomplete by now?)

DaniLazar, 2016-01-20:
Thanks for your answers!

Well, again, I am getting a code from the remote control, when reading it with ReciverDemo_Simple and it is a DECIMAL one so the other methods are not relevant in this case.

meanwhile I tried another library:
http://randysimons.com/overige/browsable/433MHz/ArduinoRemoteSwitchLibrary.7z

I don't know what he did different from rcswitch but with his functions it WORKS for me!
I used the  Show_received_code to read the code (a different decimal one then I got with rcswitch) and the I transmit it with the function  RemoteSwitch::sendTelegram(code, 11); with the code I read before and it works!

May be it will help you guys to add/fix addition supports to more chips coding?
As I don't come from the hardware areas, I will be happy to get a little explanation what is the secret here? :-) 
Is the different frequency and timing each chip codes the code?

Thanks!

fingolfin, 2016-05-05:
It's impossible for me to tell what the problem here is, as you do not provide enough details.

The DECIMAL printed by ReciverDemo_Simple should work as input for the appropriate `send()` method. Note that there are *two* of them -- one takes a string, but the other takes, well, a decimal and a length (24 in your example, based on what you write. If that didn't work, it's hard to say why with so little information, and without any ways to test it.

I am afraid looking at that library won't help. I had a brief glance at it, and the code is not very well documented, so reverse engineering it would be a pain -- and without any way to test the resulting code, it would almost certainly be wrong anyway.

closed 2016-05-05 by fingolfin