sui77/rc-switch #25: Connecting the two receivers

MeJIyA, 2016-01-17:
The problem is when you connect the two receivers(315MHz and 433MHz). In consequence of using a variable of type static, it overwrites data in a single memory controller.

fingolfin, 2016-01-17:
It's very hard to understand what you are saying.

But I am guessing you are requesting support for using two receivers simultaneously, which is currently not possible.

Unfortunately, there seems to be no "elegant" way to fix this. The problem is that `RCSwitch::handleInterrupt` has no way to know for which pin respectively interrupt it was invoked. 

Since `RCSwitch::handleInterrupt` does not know the source of the interrupt, we cannot use it to pass data to multiple RCSwitch instances.

This is a  limitation of the Arduino interrupt handling system, it seems :-/.

fingolfin, 2016-01-17:
One hackish workaround: If we arbitrarily limit ourselves to e.g. 6 interrupts, we could provie six "outer" interrupt handlers; then modify `attachInterrupt` to use the correct handler for each interrupt. Next, we'd use a small global array to lookup which RCSwitch instance is assigned to which interrupt, and dispatch to that.

But that is still limited in flexibility and somewhat ugly. Hrm.

MeJIyA, 2016-01-18:
Sorry for my English. Thanks for the answer.

sui77, 2016-01-18:
Another dirty hack that might work would be switching between both receivers once every second. Of course with the risk of missing some signals, depends on your needs.

void loop() {
   mySwitch.enableReceive(0);
   delay(1000);
   mySwitch.enableReceive(1);
   delay(1000);
}

MeJIyA, 2016-01-19:
sui77 Thank you. But I do not fit this option, you must monitor two frequencies at once. I implemented without interruption. But the accuracy lutshe wants to be, although operational.

MeJIyA, 2016-01-19:
Still managed to use the hack . delay ( 200 ) and set the tolerable results obtained .
Hedgehog Thanks again sui77.

closed 2016-02-10 by sui77

bilogic, 2016-05-22:
Is there a reason why the class uses static definition? as in RCSwitch::?
I had the same need for 2 receivers and added code for an 2nd detection buffer. I can share the code if it helps, but I thought the best way was to use a non static approach.

MeJIyA, 2016-05-23:
Thank you , I have long been solved the problem of writing separate code to the second receiver .

fingolfin, 2016-05-23:
@bilogic The class uses static definitions because Arduino interrupt handler callbacks sadly only can access static / global data; and interrupt handlers cannot easily be shared for different users. In particular, they have no  `void *refCon` parameter, which many other callback based APIs offer.

So I am really curious to learn how you bypassed that problem. But please let's not put it onto this old, closed issue; instead, submit a pull request, and/or a new issue where you explain your changes.

MeJIyA, 2016-05-23:
My level of English is low and I did not understand everything. But answer me more . I used the library on the same frequency , but dorabatival library. I added Oregon and Lacrose. I also added support for long codes for other sensor types. In the future, when I will modify the code I'll post the code . At the time of Danian though workable but not beautiful .
Another receiver potsepil I just interrupt and organized another function but to recognize only one protocol .
I just I measure the duration and analyze the code and decodes it . Though excessively but it works.

bilogic, 2016-05-23:
@fingolfin, @MeJIyA my codes and suggestions are on #63

Mshad0w, 2019-10-26:
hi,
great job...
as another hack ,if you include <PinChangeInt.h> library to RCSwitch.c,
and replace this line in  RCSwitch::enableReceive() functon:
 attachInterrupt(this->nReceiverInterrupt, handleInterrupt, CHANGE); 
with :
attachPinChangeInterrupt(this->nReceiverInterrupt, handleInterrupt, CHANGE);   
receive function can be attached to every other pins of board,by adding pin number in enableReceive(pin_number) routin...
useful for nodemcu , esp8266 and etc...

AKTanara, 2021-07-20:
I solved this problem for myself by simply duplicating the library with new name "RCSwitch2" and created the second receiver instance with that library. It worked perfectly. So if you are interested in this solution it is just enough to copy the library to another folder with name "RCSwitch2" and replace all "RCSwitch" phrases in the files with "RCSwitch2". This is most easily done by Notepad++ in few clicks.
[image]
[image]