sui77/rc-switch #26: Interfacing temperature sensor

roccomuso, 2016-01-21:
What happen if I try to detect a temperature sensor?
What rc-switch library provides to me? weird integer codes?

Matzz, 2016-01-22:
Some time ago I tried to detect temperature data using rc-switch. I didn't get anything. RC switch treated this input as noise.

gorec2005, 2016-01-23:
temperature sensors have a very different and more complicated protocols depend from vendor/maker...

HakimSMH, 2016-01-25:
I have some "Door Opening" sensors. RC-Switch doesn't get anything as well, i thought the problem was with the 32bits limitation of RC-Switch, as my sensor is 36bits. Any solution ?

roccomuso, 2016-01-25:
Have your door sensor a supported chipset? I had no trouble with a PIR sensor.

SunboX, 2016-02-03:
I have a `COCO KlikAanKlikUit AMST-606` door/window sensor. RCSwitch doesn't recognize anything (sadly). RFControl recognizes some data if I open/close the switch, but I can't use taht data out of RFControl.

SunboX, 2016-02-03:
Here's some info about the protocol of my sensor: https://github.com/pilight/pilight/blob/master/libs/pilight/protocols/433.92/arctech_contact.c

SunboX, 2016-02-03:
Found a library that works :smile: -> https://bitbucket.org/fuzzillogic/433mhzforarduino/wiki/Home (NewRemoteSwitch)

closed 2016-02-10 by sui77