sui77/rc-switch #27: Comments

HakimSMH, 2016-01-27:
The code is rather well documented, but i still have some troubles understanding the signification of the changeCount, repeatCount and the Timings array. Would be glad if you could comment them.

In addition, but it perhaps due to my weak experience, i didn't get the Protocols signification in terme of each member of the structure and How to use them. As I have some 433Mhz sensors that RCSwitch doesn't get i thought i could add in some new protocols.

Thanks

sui77, 2016-02-18:
repeatCount: A remote sends its code repeatedly as long as you press the button. A signal is only classified as received when it is received twice in a row - if (repeatCount == 2) { ... }

changeCount: RCSwitch::handleInterrupt() is called whenever the signal changes from high to low and vice versa. Have a look at https://github.com/sui77/rc-switch/wiki/KnowHow_LineCoding 
When the long low sync signal is detected ( if duration > RCSwitch::nSeparationLimit ) it will assume a new signal is starting ant thus set changeCount=0 and save all following times between high/low states in timings[changeCount++]. 
The timings array is then evaluated in RCSwitch::receiveProtocol.

HakimSMH, 2016-02-18:
Thank you ! It makes things clearer :)

closed 2016-02-18 by HakimSMH