sui77/rc-switch #32: Sending to GT-FSI-07 / GT-9000 not possible

dawie1000, 2016-02-13:
Hello

is it possible to switch Globaltronics GT-FSI-07 / GT-9000?
I tried everything to archive that but it won't work.

Capturing raw data is possible by the receiveDemo_advanced. I received packages like (e.g.):
```
Decimal: 15114215 (24Bit) Binary: 111001101001111111100111 Tri-State: not applicable PulseLength: 101 microseconds Protocol: 3
Raw data: 7200,968,564,968,568,964,564,1196,1088,440,1092,436,328,1200,1084,440,328,1200,324,1196,332,1092,432,336,1192,1092,436,1088,444,1084,436,328,1200,328,332,1196,332,1192,332,1196,328,1196,1092,432,1092,440,564,964,572,
```

I tried to send data with protocol 3, 4 & 5 and pulse lengths of 101 and 213...

Thanks for an information if using this piece of hardware is generally possible.
Please let me know if you need further information.
David

fingolfin, 2016-02-14:
This looks like it might be protocol 3. Did you try sending back exactly that binary string?

Hmm... though now that I think about it, my Brennenstuhl RCS 2044 also needed a tweak for protocol 3 to work. Try editing the rc-switch source, and apply this change:
```diff
 static const RCSwitch::Protocol PROGMEM proto[] = {
     { 350, {  1, 31 }, {  1,  3 }, {  3,  1 } },    // protocol 1
     { 650, {  1, 10 }, {  1,  2 }, {  2,  1 } },    // protocol 2
-    { 100, {  1, 71 }, {  4, 11 }, {  9,  6 } },    // protocol 3
+    { 100, { 30, 71 }, {  4, 11 }, {  9,  6 } },    // protocol 3
     { 380, {  1,  6 }, {  1,  3 }, {  3,  1 } },    // protocol 4
     { 500, {  6, 14 }, {  1,  2 }, {  2,  1 } },    // protocol 5
 };
```
That's just replacing a 1 by a 30. 

Does that help?

dawie1000, 2016-02-15:
That's it! Thank you so much for helping (spend a lot of time for testing...). 
It works perfect with that change.

fingolfin, 2016-02-16:
Glad to hear it helped.

fingolfin, 2016-02-16:
Glad to hear.

But now I wonder: On https://github.com/sui77/rc-switch/wiki/List_KnownDevices it says that this device needs protocol 4 or 5. 

But this is made by Aldi, a discounter -- so my guess is that they sell different hardware under the same name? Would you tell us when you bought it? Does it have any identification that might be used to distinguish from other outlets with the same model number?

dawie1000, 2016-02-16:
I bought that device approx. 1,5 years ago at Aldi (Süd). I don't know much about it and I also don't know, if they sell different hardware under the same name. 

On the remote is printed the following data:

> Globaltronics GmbH & Co. KG.
> Domstr. 19, 20095 Hamburg
> E47077
> Modell GT-9000
> ...
> XI/06/2014   91729   20017057

On the outlet is printed:
> Globaltronics GmbH & Co. KG.
> Domstr. 19, 20095 Hamburg
> Modell GT-FSI-07
> ...
> E47077   XI/06/2014    91729    20017057

I'm not sure if that helps - but maybe it helps other users to identify their device.
Please let me know if you have further questions.

sui77, 2016-03-01:
merged

closed 2016-03-01 by sui77

dreckbolle, 2017-11-20:
Hey @dawie1000 you wrote:
> I tried to send data with protocol 3, 4 & 5 and pulse lengths of 101 and 213
I have the same device(s), could you please send some more information about the correct (most reliable) send commands? I can receive the corresponding data from the GT-9000, but I have problems controlling the GT-FSI-07. From the obtained codes is see that my Pulse length is about 500, strikingly the receive codes seem to change every time, although I press the same button on the GT-9000. 
I think, it would also be nice if this information could be found on the wiki. I could write something about it @sui77, if it works. 
Cheers, Christoph

dawie1000, 2017-11-20:
Hi Christoph,

I use the following parameters:
`mySwitch.setProtocol(5);`
`mySwitch.setPulseLength(511);`

And keep in mind, that the GT-9000 sends out 4 different codes per button. 

Let me know, if you need some more information.

dreckbolle, 2017-11-20:
Thank you dawie1000! That was the information I was looking for. But honestly, who invents a remote control like this? Well, it works now...

dawie1000, 2017-11-21:
Yeah, very interesting behavior... Took me some time, too.
Good that it works now!

goldiiii, 2018-01-05:
Hi all. I am also trying to get this work. My problem is .. I only have a raspberry pi and so i think i cannot use the  receiveDemo_advanced.

I could only get this from my transmitter:
Received 10223472 / 24 bit Protocol: 5
Received 9910288 / 24 bit Protocol: 5
Received 9576116 / 24 bit Protocol: 5
Received 9865508 / 24 bit Protocol: 5
Received 9576124 / 24 bit Protocol: 5
Received 9865516 / 24 bit Protocol: 5
Received 10470802 / 24 bit Protocol: 5
Received 10251074 / 24 bit Protocol: 5

Does this help me to get it working?