sui77/rc-switch #34: RC-Switch Library on M0 Pro Problem with delaymicroseconds()

thkaisin, 2016-02-18:
[test_RCSwitch_on_M0.txt](https://github.com/sui77/rc-switch/files/136846/test_RCSwitch_on_M0.txt)

Hello,

Trying to use RCSwitch library on M0 Pro, timing of delaymicroseconds() is changed.
Measuring pulse length with scope: In the simple attached example, timing of delayMicroseconds(100) changes from 100uS to >600uS, 200 uS => 1,4mS after switchOn() is called.
No impact on delay() function.

Thank you for your support.

fingolfin, 2016-04-20:
So, in other words delayMicroseconds() is horribly broken on M0 Pro?

Sounds like a bug you should report to the M0 Pro folks... In this particular case, googling for "delayMicroseconds
quickly leads to <https://github.com/arduino-org/Arduino/releases/tag/1.7.4>, which indicates that a bug in delayMicroseconds -- so they know about this bug, and fixed it, but it's not in the 1.6.x series.

You could try using Arduino 1.7.9 (the latest, at the time I write this) to see if it helps.

But there is nothing we can do about it from our side, sorry.

closed 2016-04-20 by fingolfin

thkaisin, 2016-04-20:
Hello,

You are right, bug in delayMicroseconds(), I solved by doing another
delay_Microseconds() and using in RCSwitch() and everything works perfectly:
void delay_Microseconds(int time)
{
  int End;
  End = micros() + time - 4;
  while (micros() < End){ ; }
}

Thank you for your reply.