sui77/rc-switch #36: ReceiveDemo Simple and esp8266

yomasa, 2016-03-02:
Does anybody know if he enableReceive(0) is pin 2  on the ESP 8266, Will this work or is it not compatible. Are there maps interrupts to pins on the ESP?  I get to load without any errors; just not receiving anything?

sanych1by, 2016-03-14:
It's work. But number of pin is direct connection, like Gpio4 = 4. So if you connect data pin of reciever to gpio2 use enanleRecieve(2). Also remember esp 3.3v, not 5v and usuall vcc of 433 modules is 5v. So you need connect data pin using resistors –> data-20kOm-gpio2-10kOm-GND.

closed 2016-03-31 by sui77

cyppressmx, 2018-03-03:
it doesn't work for me

i use next sketch in Arduino uno and Mega, but i don't know how use it in nodeMcu, can anyone help me?!

#include <RCSwitch.h>
RCSwitch mySwitch = RCSwitch();
void setup()
{
  pinMode(13,OUTPUT);
  //pinMode(LED_BUILTIN,OUTPUT);
  Serial.begin(9600);
  mySwitch.enableReceive(0);  // pin #2 for arduino uno, Pin #0 for Mega
}
void loop()
{
  Serial.print(Serial.read());
  if (mySwitch.available())
  {
    long  int value = mySwitch.getReceivedValue();//en la libreria viene int
    if (value == 0) 
    {
      Serial.println("error de codigo");
    } 
    else 
    {
      Serial.print("recive code: ");
      Serial.println(value);
     Serial.println( mySwitch.getReceivedValue() );
    }
  }else{
     mySwitch.resetAvailable();
  }
}