sui77/rc-switch #42: HomeEasy support

DirkHeinke, 2016-03-31:
Hi,

do you plan to add Home Easy support with the advanced protocol? Some information about the protocol can be found here: http://playground.arduino.cc/Code/HomeEasy

fingolfin, 2016-05-04:
Well, this library is mostly about supporting switches, not more complicated stuff... but I see no fundamental reason why not to add support for that, if it fits in sensible.

But to write such code, one needs a HomeEasy device in order to test and debug it. So I certainly won't do it.

But if somebody (perhaps you?) sends a patch or tells us otherwise what changes to make, we will definitetly consider it. In that case, please submit a pull request, or if that is beyond your tech skills, an issue with the code.

Note: on a quick glance, it seems that the low level encoding matches our protocol 1 ("pulses are expected to be either 375us or 1125us" is close enough to 1*350us and 3*350 = 1050us). Perhaps one of the existing switchOn() / switchOff() or send() method can already be used?

gmarco, 2016-11-24:
I have a Homeeasy. Sendign worked after setting pulse width to 325 andsend with mySwitch.switchOn('d',2);

Hope this helps

closed 2016-12-22 by sui77

jymbob, 2017-06-28:
The comment above appears to be using the HomeEasy 200 system, which sends a 'house code'(A to P) and a unit number.
I'm looking into this as someone with the newer HomeEasy switches (HRR3). This uses a 26bit 'id' code, a 'group' bit, an 'on/off' bit and a four bit button id (controlling up to 16 sockets with one remote).
The RCSwitch receive code doesn't pick up anything from the remote, but using another library I've managed to determine the relevant codes.

So, the principle appears to be to send a latch code, a 32bit code as manchester pairs, then a second latch code.
The final gotcha is that the sockets don't seem to respond reliably unless you send the code twice.

Hope this is helpful. I'm afraid I don't know enough about how RCSwitch works to convert that into useful code, but am happy to test if someone else can!

jrc, 2020-07-24:
Duplicate of #124?