sui77/rc-switch #44: Out of index assign of value

WoodsterDK, 2016-04-13:
Hi there,

I think there might be an out of index assignment here : 

diff(RCSwitch::timings[i + 1], delay * pro.zero.low) < delayTolerance) {

Perhaps this can be solved by adding:
	if (changeCount < RCSWITCH_MAX_CHANGES)
	{

fingolfin, 2016-05-02:
To be precise, this is a read, not an assignment, but that doesn't matter, out of bounds is out of bounds.

Now, let's see: `RCSwitch::timings[i + 1]` may end up accessing `RCSwitch::timings[changeCount]`. Moreover, the largest value `changeCount` attains is `RCSWITCH_MAX_CHANGES`, thanks to the check in `RCSwitch::handleInterrupt()`.

Then, right before `receiveProtocol` is invoked, `changeCount` gets decremented. Thus, the maximal value for `changeCount` we ever see in `receiveProtocol()` is `RCSWITCH_MAX_CHANGES-1`. 

The maximal value `i` can attain is `changeCount-1`, thus `RCSWITCH_MAX_CHANGES-2`. But that suggests to me that the access above actually is always correct.

Am I missing something? If not, then there is no bug. But regardless of that, I think this code is too convoluted for its own good, and should be cleaned up.

WoodsterDK, 2016-05-03:
Well you might be right. I was having a lot periodically problems on a ESP8266 and began looking for possible causes. 
Most likely those issues had something to do with the interrupt not being placed in ram as suggested elsewhere. Thanks for your time and effort.

closed 2016-05-03 by WoodsterDK