sui77/rc-switch #45: ENC28J60

inflamers, 2016-04-22:
Hello

I have a small project running on an arduino Uno, when receiving a 433Mz signal from a sensor, the arduino sends an email through ENC28J60 module. Everything almost works, once it's connected to the internet, the initialization of the module makes my arduino to reboot continuously. I discovered that the issue comes from the function mySwitch.enableReceive(0). For some reason, using this interrupt is the root cause. My question is, is there any other solution to use this function with pin 2 and no interruption ?

Thanks !
Mickael.

fingolfin, 2016-04-23:
In the current implementation, I am afraid some pin with interrupt capabilities is required. For the Uno, that means pin 2 or pin 3.

But if you actually have pin 2 available for use with rc-swich, I don't quite see what the problem is / what is causing it. Weird...

inflamers, 2016-04-23:
It seems I can't have UIPEthernet with ENC28J60 module and RCSwitch running at the same time...

inflamers, 2016-04-25:
Basically, here is below a piece of my code, I have 3 main functions in setup() :
- mySwitch.enableReceive(0);
- server.begin();

`Serial.println("Starting rcswitch");
  mySwitch.enableReceive(0);
  Serial.println("Starting comm");
  Ethernet.begin(mac);
  server.begin();`

When I execute this, I have an infinite loop in the monitor :
Starting rcswitch
Starting comm
Starting rcswitch
Starting comm
Starting rcswitch
Starting comm
...

If I disable mySwitch.enableReceive(0); with comments ("//") everything else works perfectly fine, but not rcswitch.Any idea ? Pretty weird. I used pin 2 for RCSwitch (Interr. 0).

Mickael

fingolfin, 2016-05-03:
Perhaps something else is also trying to use pin 2 and/or messing with interrupts? E.g. the ethernet code?

In any case, I doubt this is due to a bug in rc-switch. You are most likely better of asking for help in a general Arduino forum. And if you could provide more (or ideally all) of the source code, it would be easier to pinpoint where things go wrong.

inflamers, 2016-05-03:
This is what I was thinking, I guess the issue comes from the ethernet module. I already opened a thread in the general forum and in the github page for ENC28J60, no solution so far... Thanks to you anyway !

fingolfin, 2016-05-03:
Actually, what library are you using to drive the ENC28J60? AS far as google tells me, the stock ethernet library is not compatible with it.

fingolfin, 2016-05-04:
Without more details (such as what library you use to drive the ENC28J60; or perhaps a complete "minimal crashing demo"), there is nothing really we can do about this.

closed 2016-05-04 by fingolfin