sui77/rc-switch #50: Problem detecting a single frame,  sync start condition.

WoodsterDK, 2016-05-03:
Hi I have used the transmit function to learn a copy remote some values. It seems the receive part only works if multiple values are sent (long press on my remote makes it retransmit ) . Could there be something with an issue getting in sync.

fingolfin, 2016-05-04:
I think the receive code requires the signal to be transmitted twice before it accepts it. Not sure why, but you could try changing the line which says `if (repeatCount == 2)` to simply `if (true)` -- does that help?

@sui77 Any reason why the code waits for two full transmissions before accepting the signal?

fingolfin, 2016-05-04:
Note that for my switches, no matter how short I press, I get at least two full repetitions of the signal. For another remote, it's even never below 5 repetitions.

rotv, 2016-05-04:
I assume it's just to avoid false matches. My remotes also sends the code multiple times.

WoodsterDK, 2016-05-13:
Ok, Will try it.

closed 2016-05-13 by WoodsterDK