sui77/rc-switch #51: Can't compile latest?

rotv, 2016-05-04:
Just me?

~/utils/433Utils/RPi_utils $ make
g++ -DRPI   -c -o ../rc-switch/RCSwitch.o ../rc-switch/RCSwitch.cpp
../rc-switch/RCSwitch.cpp: In member function ‘void RCSwitch::send(long unsigned int, unsigned int)’:
../rc-switch/RCSwitch.cpp:493:26: error: ‘bitRead’ was not declared in this scope
<builtin>: recipe for target '../rc-switch/RCSwitch.o' failed
make: *** [../rc-switch/RCSwitch.o] Error 1

fingolfin, 2016-05-04:
bitRead is part of the Arduino libs. I only used it for extra readability, but will get rid of it.

fingolfin, 2016-05-04:
I hope 1906097da60b7df5d6c40233fa7c37db1a48b0e6 resolves this

rotv, 2016-05-04:
Now it compiles but I can't send or receive RF signals. Investigating.

rotv, 2016-05-04:
I just verifier that it works with my old fork, so something has taken the turn for the worse recently :(

rotv, 2016-05-04:
It works all the way up to commit 90d00c1

rotv, 2016-05-04:
It seems like the refactor of the send code af0c25a has broken it (I tried by checking out ad0c25a and manually fixed according to 1906097).

rotv, 2016-05-04:
Got it! The for loop at RCSwitch.cpp:478 sends the bit in the wrong order.

rotv, 2016-05-04:
Closing this issue (the compilation error is solved by1906097). Opening a new one for the issue of incorrect bit order: #52.

closed 2016-05-04 by rotv