sui77/rc-switch #54: Doesn't work with the dooya dc90

coollorenzo, 2016-05-08:
I recently started with connecting my rf devices to the internet using an arduino to copy the rf (decimal)singals and using a raspberry pi to send the singals. My rf light switches work great! but when I trie to copy the singal of my sun screen which uses the dooya dc90 remote([fcc](https://fccid.io/VYY-DC90))  I get nothing I don't know why because the [RF report](https://fccid.io/document.php?id=2706979) says it uses 433.92MHz as frequency and ASK which my as modulation  receiver supports Do you guys have any ideas on why I can't receive the singals?

fingolfin, 2016-05-23:
The frequency 433Mhz is used for many different kinds of signal transmission. So there are many such signals (or rather: signal encoding) which RCSwitch does not know about and hence cannot decode.

That said, it usually isn't too hard to add support for them to RCSwitch. But for that, you need to determine some details about the signals, and then from these guess the right "protocol" settings.

I have some sample code for doign that, but it's old and hasn't been tested for a long time. A better way to help you and the many other people with similar problem might be to add a "Raw Receiver" demo, and/or a "raw" receiver mode to RCSwitch: This mode basically would return the raw timings, not trying to interpret them.

I'll try to whip something up, but probably won't get to it before the weekend. If somebody else has code readily availble which reports the timings @coollorenzo needs, please don't wait for me :-).

See also isue #20

KoenWijnstok, 2016-05-23:
I would really love if RCSwitch would at least be able to sniff and repeat the sniffed code. I think it would definitely be a big improvement for RCSwitch as you can use it to use the more obscure plugs.

fingolfin, 2016-05-24:
The FCC link you provide contains schematics, which indicate that the remote uses a MC 263 BN IC -- but I could not find any information on this. So in the meantime, you'll have to experiment yourself, e.g. by running test code as on #60 (the user there provided the C code he used to capture data from this remote).

frantona, 2017-01-03:
Hi, where you able to manage it? I have the same issue when trying with the 1660 Dooya remote control form y shades. Thanks!

coollorenzo, 2017-01-03:
Sorry but I haven't had time for it yet because of the restructuring of my iot setup but whenI get it to work I will let you know

joshlacal, 2017-08-17:
hey @coollorenzo @frantona, I've got a Dooya DC90 too, have you managed to get it working?

joshlacal, 2017-08-24:
For anyone who happens to stumble on this, I was able to control my motor (which is controlled by a DC90 remote) with this: https://github.com/bjwelker/Raspi-Rollo

They use a modified version of rc-switch, which I think added support for quad state bits. Using an Arduino and their provided code, I was able to sniff the codes from my DC90 remote. And then, with a Raspberry Pi, I used their modified rc-switch code (the sendv2 program) to send the codes to the motor.

jotadepicas, 2017-09-01:
@joshlacal @coollorenzo Hey, I can confirm that https://github.com/bjwelker/Raspi-Rollo worked for me as well! I was getting nothing out of rc-switch using the provided examples to sniff my DC90 remote, but with raspi-rollo arduino sketch (https://github.com/bjwelker/Raspi-Rollo/blob/master/Arduino/Rollo_Code_Receiver/Rollo_Code_Receiver.ino) I was able to sniff the codes :smiley:

jotadepicas, 2017-09-01:
Hmm, however, it seems I cannot send the codes to the motor. The codes I sniffed look exactly like the ones in http://physudo.blogspot.com.ar/2013/08/home-automation-mit-dem-arduino-und-433_18.html (which is the blog post referenced by Raspi-Rollo). That makes me think the codes were captured correctly.

Now for some reason, I'm sending those exact same codes but nothing happens... :/
I can receive the sent codes by using the 433 receiver used to capture them in the first place. I think that indicates the codes are being sent correctly. But I suspect I'm missing some sync signal or something, which causes the motor to ignore the codes I'm sending.

I'm not using the sendv2 program, but my own version for aduino. However I'm using the same library in the same manner so I don't think that's the issue. My code looks like this:

```
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(9600);
  Serial.println("Starting...");
  mySwitch.enableTransmit(12);
  mySwitch.setProtocol(4);
}

void loop() {
  Serial.println("Up...");
  
  mySwitch.sendQuadState("0F0F00FQ00FQQ10F0F0F");
  mySwitch.sendQuadState("0F0F00FQ00FQQ10F0F1Q");
  
  delay(2000);
  
  Serial.println("Stop...");
  mySwitch.sendQuadState("0F0F00FQ00FQQ10FFFFF");
  
  delay(2000);
  
  Serial.println("Down...");
  
  mySwitch.sendQuadState("0F0F00FQ00FQQ10F0101");
  mySwitch.sendQuadState("0F0F00FQ00FQQ10F0110");

  delay(2000);
    
  Serial.println("Stop...");
  mySwitch.sendQuadState("0F0F00FQ00FQQ10FFFFF");
  
  delay(2000);
}
```

Any ideas? @joshlacal did the sending part work out of the box for you? Thanks!

joshlacal, 2017-09-01:
Yes it did work out of the box for me, but I only used a Raspberry Pi to send the codes... I used the Arduino Rollo_Code_Receiver.ino code to sniff the codes, and then I switched to my Raspberry Pi and used the sendv2 program to send the codes. Do you have a Raspberry Pi you can use to test with?

To be honest, I am not that familiar with Arduino programming (I actually bought one just so I could try the Rollo_Code_Receiver.ino to sniff the codes), but I think (I am not sure) the RCSwitch libraries for Raspberry Pi and Arduino are slightly different. You might be trying to use the Raspberry Pi/Raspi-Rollo version of RCSwitch on an Arduino, and that could be why it isn't working? You might want to try comparing the Arduino and Raspberry Pi rc-switch libraries and modifying the Raspi-Rollo rc-switch (which supports quad state) to work with Arduino, or try porting over the quad state code from Raspi-Rollo to the Arduino rc-switch library. I'm not sure exactly what you would need to port over or if it would work, but it's a guess. 

Also, off-topic from RC-Switch, but if you are using an Arduino Mega, you could also try this firmware: http://www.rflink.nl/blog2/ I haven't gotten a chance to try it yet, but I would like to since it seems to have better integration with the home automation software I'm using (Home Assistant). DC90 support is listed under the "Brel motor", and I googled it and it seems to be the exact same remote: https://www.brel-motors.nl/webshop/besturingen/zenders/detail/166/dc-90--handzender-1-kanaals.html

jotadepicas, 2017-09-01:
@joshlacal thank you for your comments! Hmm, I assumed the rc-library was the same. In fact what I did was take an [old arduino version from 2012](https://github.com/sui77/rc-switch/tree/e89d35346a51c549ef175a8010d48063f163079b) (as stated in [this recent comment](http://physudo.blogspot.com.ar/2013/08/home-automation-mit-dem-arduino-und-433_18.html?showComment=1494968522823#c195116435676571193)) and applying the changes described in that old post, and it worked. I could communicate my transmitter and my receiver.

But, it could be there's some small difference I'm missing. I do have a raspberry pi, so I'll try directly with the pi and see what happends. I don't have an Arduino Mega, but I think this is a good excuse to get my hands on one, hehe.

I'll try this different options and hopefully I'll report back with some degree of success :stuck_out_tongue:

jotadepicas, 2017-09-02:
@joshlacal today I got an Arduino Mega, flushed the rflink firmware and... ta-da! it works! I can both sniff commands and retransmit them to the motor! Thank you so much for the tip :smile: 
Now, I don't know exactly why it wasn't working with the modified rc-switch version, but I think maybe we can find out by looking at the "BrelMotor" protocol implementation in rflink firmware.

Here are some details just in case anyone else stumbles upon this:

- I'm using an Arduino Mega 2560 16au
- Downloaded latest rflink firmare from here: http://www.rflink.nl/blog2/download
- rflink comes with a loader program for windows. I'm using ubuntu so I used `avrdude` command line tool to install the firmware into de arduino: `avrdude -v -p atmega2560 -c wiring -P /dev/ttyACM0 -b 115200 -D -U flash:w:RFLink.ino.hex:i`
- Wired my RF 433 Mhz transmitter/receiver kit as pointed out here: http://www.rflink.nl/blog2/wiring (basically, pins D14, D15 and Ground for the transmitter, and pins D16, D19 and Ground for the receiver, but have a look at that page for instructions on each particular RF module).
- Once everything was set up, I opened Arduino IDE and used the Serial Monitor at 57600 baud, with "Newline" setting for ending sent lines, and I could communicate sucessfully with the gateway, as described here: http://www.rflink.nl/blog2/protref (note the newline setting and correct baud speed, otherwise communication will fail).

I pressed the buttons on my remote and in the Serial Monitor console I got something like:

```
20;03;BrelMotor;ID=a12312;SWITCH=b1;CMD=UP;
20;04;BrelMotor;ID=a12312;SWITCH=b1;CMD=STOP;
20;05;BrelMotor;ID=a12312;SWITCH=b1;CMD=DOWN;
20;01;BrelMotor;ID=a12312;SWITCH=b1;CMD=STOP;
```
So, I sent the following comands and it worked like a charm:

```
10;BrelMotor;a12312;b1;UP
10;BrelMotor;a12312;b1;STOP
10;BrelMotor;a12312;b1;DOWN
10;BrelMotor;a12312;b1;STOP
```
As I mentioned before, maybe this "BrelMotor" protocol in rflink can give us a clue on how to properly support it on rc-switch :thinking:

frantona, 2017-09-25:
Hi! Does anyone have a step by step process for this?? I followed RC-Switch step by step before, but it didnt worked, so I hope this Rollo way works. Sorry to ask, Im a big Noob on this.

testermax, 2018-05-10:
Hi,
Im using an nodemcu v3, and the struggling to sniff my dc90 with the "Rollo_Code_Receiver" compiles and runs fine.
But I cant seems to recieve anything when I press any of the buttons on the dc90.
I have tested for hours with the RCSwitch library, but with no luck. I was hoping that I could manage to send commands from my nodemcu, just like the dc90 does.
Does anyone have a clue?

testermax, 2018-05-10:
Hi. I have struggled with getting rcswitch to work with nodemcu v3, with no luck. I have download the example code, and tested all gpios. I have also verified my reciever. Nothing seems to work and the serial view will not show any results, when clicking my dc90 remote.
Does anyone have a clue?

Martin-Laclaustra, 2018-05-12:
https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_1

huexpub, 2018-05-15:
https://forum.mysensors.org/uploads/upload-17387506-e377-47f3-b11c-4674aeac6d0a.pdf
e, wait Okami do it for rc-switch

testermax, 2018-06-02:
Hi,
I kinda gave up on reading the signals from my dooya DC90 remote, with XD-RF-5V reciever. I was able to sniff and decode the signal with SDRSharp instead.
But have anyone experienced problem with the library when transmitting over 433 Mhz with the FS1000A?
Im able to transmit over 315 Mhz, but its totally "radio silence" on the 433-434 Mhz range.
I have tested with two different FS1000A modules, with/without 17 cm antenna.
The reason I mention it here, is Im not sure if this is a hardware thing, og a library thing.

testermax, 2018-06-03:
I have now decoded my remote, built my application around the rcswitch library, to recieve commands from either MQTT or blynk. My plan is to forward these command to the rcswitch library and the 433mhz version of FS1000A. My challenge is that Im not sure if this library is the correct one to use?
I have attached the recorded signals from the awning "out" command. What do I need to do to send these commands correct? Can I just use the send("11001010101....") function, or do I need to calculate the timing? I would appreciate som guideline on how to get the signals right.
[image]
[image]
[image]

ChrisBall, 2019-07-30:
re: quad state receiving and sending on Dooya remotes:

Thanks everyone for your comments above. Not sure if this an appropriate place to post, but this thread helped me through my issues in just a few hours instead of days. Adding further info for anyone else that finds themselves here.

To confirm, I was able to use the arduino program here: [https://github.com/bjwelker/Raspi-Rollo](https://github.com/bjwelker/Raspi-Rollo) to sniff the quad-state codes for my device (Dooya DC1603) using an arduino UNO and [MX-05 receiver](https://forum.hobbycomponents.com//viewtopic.php?f=74&t=1324).

These are similar, but not identical to the codes given here: [http://physudo.blogspot.com.ar/2013/08/home-automation-mit-dem-arduino-und-433_18.html ](http://physudo.blogspot.com.ar/2013/08/home-automation-mit-dem-arduino-und-433_18.html )

e.g. mine are:
```
char UPSYNC[]=    "F000F0F00Q0FQ10F0F0F";
char UP[]=        "F000F0F00Q0FQ10F0F1Q";
char DOWNSYNC[]=  "F000F0F00Q0FQ10F0101";
char DOWN[]=      "F000F0F00Q0FQ10F0110";
char STOP[]=      "F000F0F00Q0FQ10FFFFF";
```

It would seem only the last 4 quadbits designate function, and the first 16 will vary between remotes/remote models. The remote repeats each code 5 times.

I was then able to send the codes using the same arduino UNO, [a MX-FS-03V Transmitter](https://hobbycomponents.com/wired-wireless/617-433mhz-wireless-transmitter-module-mx-fs-03v), and the modified rcswitch library found here: [https://github.com/bjwelker/Raspi-Rollo](https://github.com/bjwelker/Raspi-Rollo).

Code:
```
#include <RCSwitch.h>  //modified quad-state version of the library found here: https://github.com/bjwelker/Raspi-Rollo](https://github.com/bjwelker/Raspi-Rollo

RCSwitch mySwitch = RCSwitch();

char UPSYNC[]=    "F000F0F00Q0FQ10F0F0F";
char UP[]=        "F000F0F00Q0FQ10F0F1Q";
char DOWNSYNC[]=  "F000F0F00Q0FQ10F0101";
char DOWN[]=      "F000F0F00Q0FQ10F0110";
char STOP[]=      "F000F0F00Q0FQ10FFFFF";

void setup() {
  Serial.begin(115200);
  mySwitch.enableTransmit(10);
  mySwitch.setProtocol(4);
  mySwitch.setRepeatTransmit(5);
}

void loop() {
  Serial.println("Sending UP");
  mySwitch.sendQuadState(UPSYNC); //UP SYNC
  mySwitch.sendQuadState(UP); //UP
  Serial.println("Sent");
  delay(5000);
  
  Serial.println("Sending STOP");
  mySwitch.sendQuadState(STOP); //STOP
  Serial.println("Sent");
  delay(2000);
  
  Serial.println("Sending DOWN");
  mySwitch.sendQuadState(DOWNSYNC); //DOWN SYNC
  mySwitch.sendQuadState(DOWN); //DOWN
  Serial.println("Sent");
  delay(5000);
  
  Serial.println("Sending STOP");
  mySwitch.sendQuadState(STOP); //STOP
  Serial.println("Sent");
  delay(2000);
}
```

elbusa2020, 2021-06-20:
Hello everyone, did you manage to solve this problem with the Dooya DC90 controls? I have been doing tests, I can sniff although the TX does not respond, engine does not respond. I await your comments. Greetings

gianricod, 2022-06-12:
thanks you guys. all that was very useful.

i had to make a modified version of https://github.com/bjwelker/Raspi-Rollo/tree/master/rcswitch-pi  because i have a nanopim4v2 using armbian and i had a lot of issues making wiringpi work 

My hack consists in the use of libgpiod rather than wiringpi.
Sharing here in case it could be useful to someone else: https://github.com/gianricod/dooya-roller-nanopim4v2
By the way please note that the gpiochip is hardcoded in RCSwitch.cpp, so you need to find yours.