sui77/rc-switch #56: ReceiveDemo_Advance - Compiler error IDE 1.6.6

bdonaubauer, 2016-05-12:
Hello Board,

after compiling the Advance Demo i got the following compiler error

exit status 1
'char* dec2binWzerofill(long unsigned int, unsigned int)' was declared 'extern' and later 'static' [-fpermissive]

I have tried the actual IDE version 1.6.11 and the older version 1.6.2.

Thanks for helping

bau-sec, 2016-05-13:
Does anyone know the status of the Travis CI builds?  The build badge in the readme links to a page saying this project isn't active. There have been a couple compile errors come up as issues recently (this one and #51) which CI builds could be catching.

fingolfin, 2016-05-23:
@bau-sec Hmm, seems to be broken. To be honest, I never was even aware it existed in the first place... Aha, I see, it was merged as part of @38 by @sui77 but I don't know whether he ever proceeded to actually activate it on the Travis side... ?

@sui77 I think you will have to become active to fix the Travis problem. At least I don't think I have the permissions to deal with it. I just tried, but Travis told me `There was an error while trying to activate the repository`. My guess is that I cannot "activate" things as I don't own this repository. Which makes sense, too.

fingolfin, 2016-05-23:
@bdonaubauer I made a change to the examples in 82c12a092a255b08cc64bd0b7e1bba7051ee8f86 which should fix this. You could also try to simply remove the `static` in front of the code for `dec2binWzerofill`

bau-sec, 2016-05-28:
This can be closed. My PR #67 added compilation of this example using the Arduino IDE and the test is passing.

closed 2016-05-29 by fingolfin