sui77/rc-switch #57: Change the transport protocol.

AHTOXA84, 2016-05-14:
Hello. You can change the transfer protocol. How to do it? Thank you.

rotv, 2016-05-14:
In code, use one of:
    void setProtocol(Protocol protocol);
    void setProtocol(int nProtocol);
    void setProtocol(int nProtocol, int nPulseLength);

Is that what you're asking for?

AHTOXA84, 2016-05-14:
My protocol:
pulselength = 320;
Sync bit = 36 low pulses and 1 high pulses;
"0" bit = 1 low pulse and 2 high pulses;
"1" bit = 2 low pulses and 1 high pulse;
key = 24 bit

AHTOXA84, 2016-05-22:
Prompt me how to change the protocol for this. Thank you.

fingolfin, 2016-05-23:
You can define your own instance of the `Protocol` struct with the parameters you just described, and then use the first of the three `setProtocol` methods mentioned by @rotv

AHTOXA84, 2016-05-23:
I need to change the cpp file?

fingolfin, 2016-05-24:
Wait, in your protocol, the low pulses are *first*, the high pulses second?

Then you need something like #49, I am afraid, as RCSwitch does not yet support this out of the box.