sui77/rc-switch #58: Promax support

KoenWijnstok, 2016-05-15:
Hi,

I bought a new Promax switch from our local Action store in The Netherlands. They are "self learning", but don't appear to listen to the normal protocols. When using the ReceiveDemo it looks normal, except after the Tri-State decoding there are some extra symbols (see attached file).

The receiverdemo works fine when listening to controls sent by the Remote Control, except when you transmit the same decimal number (or binary) the sockets just won't respond.
Is there any way I can make this working? I'm not the only one, it looks like a lot of people are having issues with the new sockets from the Action.

[image]

fingolfin, 2016-05-23:
The extra symbols are weird -- it looks like the string returned by `bin2tristate` was not properly null terminated.

Anyway: How do you transmit the codes, exactly? Using RCSwitch, or using custom code? If using RCSwitch, how exactly?

S-Knop, 2016-10-05:
Hi, 

I've it currently working with a small work around.,
the remote sends multiple signals, 24 bit and 28 bit. 
I've tried all buttons on the remote and multiple tools/ lib to sniff but without success.

when i send a from the remote it looks the same as yours but without those weird symbols at tristate:

**signal from remote:**
**Decimal:** 1048853
**(24Bit) Binary:** 000100000000000100010101
**Tri-State:** 0F00000F0FFF 
**PulseLength:** 314 microseconds
 **Protocol:** 1
**Raw data**: 1103337,328,31,2170,198,46,5,8,6,7,390,254,1030,254,1035,249,1035,249,1031,253,1030,254,1036,248,1049,235,1032,253,1028,256,1035,249,1030,894,393,251,1035,249,1030,254,1038,887,393,251,1033,891,395,249,1034,

then I came across a post on a dutch forum with these codes for the Promax switch set, 
when these codes are transmitted with pimatic it gives different results in RCswitch.

**transmitted with pimatic**
**pulseCount** 50
**pule length:** 222, 438, 952, 5028
0202210202212102210202022102022121210221022102212121210203 
0202210202210221022102212102212121020202022121212102210203 

**received with RCswitch**
**Decimal:** 40410462 
**(28Bit) Binary:** 0010011010001001110101011110 
**Tri-State:** not applicable 
**PulseLength:** 504 microseconds 
**Protocol:** 2
**Raw data:** 5055,225,968,222,968,953,454,222,968,223,967,953,453,953,455,221,969,953,454,222,968,223,967,223,968,952,454,222,971,225,971,954,454,960,458,955,456,225,971,954,457,229,971,954,459,226,970,953,454,953,455,953,454,952,456,221,970,

put a unit in learning mode an try the decimals from the code.
code I used.

```
/*
  Example for different sending methods
  
  https://github.com/sui77/rc-switch/

some extra decimals
A 40025046 ON
A 38856910 OFF
B 41567385 ON
B 39916253 OFF
C 240760633 ONN
C 240257549 OFF
D 105444762 ONN
D 107879350 OFF
  
*/

#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {

  Serial.begin(9600);
  
  // Transmitter is connected to Arduino Pin #10  
  mySwitch.enableTransmit(10);

  // Optional set pulse length.
  mySwitch.setPulseLength(470);
  
  // Optional set protocol (default is 1, will work for most outlets)
  mySwitch.setProtocol(2);
  
  // Optional set number of transmission repetitions.
  // mySwitch.setRepeatTransmit(15);
  
}

void loop() {

  
  /* Same switch as above, but using decimal code */
  mySwitch.send(40410462, 28); // turn A on
  delay(1000);  
  mySwitch.send(39172218, 28); // turn A off
  delay(1000);  

  delay(20000);
}
```

S-Knop, 2016-10-05:
Nvm with latest version i am receiving the propper decimals 👍  so it does support promax.

pasibun, 2017-02-07:
Hey S Knop / KoenWijnstok, 

Im currently working on the same thing. Connecting a Pi3 through RC Switch with a cheap china 433mhz TX/RX. Also bought the Action PROMAX (FA500S). 

When using the Receive_Advanced demo Im getting good results. 
What do I need to configure to send proper code to the Promax?