sui77/rc-switch #61: Travis integration not working

fingolfin, 2016-05-23:
As reported by @bau-sec in a comment on issue #56, the Travis integration (merged in PR #38) is currently not working.

I think @sui77 will have to do something on the travis side to fix this; perhaps going to <https://travis-ci.org/sui77/rc-switch> and clicking the activation link will be enough?

sui77, 2016-05-25:
Done. 

Unfortunately the ReceiveDemo_Advanced fails complaining about undeclared functions altough they are in the same file ( examples/ReceiveDemo_Advanced/output.ino ). 
Compiles fine in my Arudino IDE. Any ideas?

/tmp/tmpTH72Bq/src/output.ino:6:53: error: 'dec2binWzerofill' was not declared in this scope
/tmp/tmpTH72Bq/src/output.ino:14:34: error: 'bin2tristate' was not declared in this scope

(more: https://travis-ci.org/sui77/rc-switch/jobs/132872644 )

sui77, 2016-05-26:
Moved the two functions to top. Altough I'm not that happy to have the helper-functions at top it works now.

sui77, 2016-05-26:
Reverted and fixed with https://github.com/sui77/rc-switch/pull/66 which is even better.

closed 2016-05-26 by sui77