sui77/rc-switch #62: Record the first half of the sync signal, too

fingolfin, 2016-05-23:
Right now, our received timings omit half of the sync signal. While this is not a problem for normal decoding, it is quite annoying when one wants to analyze a new protocol, for it makes it difficult to determine the length of the first half of the sync signal; and that is important for sending signals. (See also issue #53.)

Implementing this shouldn't be hard, but I don't want to try it without access to my hardware, so that I can conduct tests :-).

bau-sec, 2016-05-23:
Minor clarification: It's not simply that half is missing; it's that the receive code ensures that the timing array starts with a long pulse. If the sync signal starts with a short pulse, the starting pulse will be omitted. If the sync signal starts with a long pulse the entire sync signal will be captured.  My comments in PR #49 illustrate and explain this in the case of a protocol in which the sync starts with a long pulse.

fingolfin, 2016-05-24:
You are of course absolutely right, thanks for pointing this out.

Martin-Laclaustra, 2016-11-07:
One more comment: The sort half (if starts with a sort half) of the sync signal is the last timing recorded.
The next timing, which the library uses to understand that the code has finished is the one that is similar to the initial long one.
So the data for that signal is already there and available in:
`MySwitch.getReceivedRawdata()[2*MySwitch.getReceivedBitlength()+1]`