sui77/rc-switch #63: Two receivers + why static definitions?

bilogic, 2016-05-23:
> @bilogic The class uses static definitions because Arduino interrupt handler callbacks sadly only can access static / global data; and interrupt handlers cannot easily be shared for different users. In particular, they have no void *refCon parameter, which many other callback based APIs offer.
> 
> So I am really curious to learn how you bypassed that problem. But please let's not put it onto this old, closed issue; instead, submit a pull request, and/or a new issue where you explain your changes.

@fingolfin 

1. Sorry, but i have no idea how to do a pull request without forking, so I'm just going to attach my code here
2. The code is a proof of concept, not too dirty, but can be much cleaner
3. I'm experienced in C/C++ but very new to Arduino
4. Initially I wanted to go the way of non static, but after reviewing the code, it was much easier to just add another buffer and interrupt handler
5. No offence, but I wasn't convinced about the "access static" data and did a test, you can find it as ```non_static``` in my code. Interrupt handler is able to read + write to it, is my test correct?
6. Assuming my test is correct (aka interrupt handlers can access non static variables), I think the best way is to have 2 instances of rc-switch, 1 for each receiver, keeping as much code and variables static while the buffers non static
7. My earlier encounter with static definitions on Arduino was to prevent memory fragmentation, this was certainly a valid consideration
8. I'm not sure if 2 or more instances will lead to any memory fragmentation or other issues

Finally, thanks for this library, I having a fantastic time turning on/off electricity :)

```
/*
  RCSwitch - Arduino libary for remote control outlet switches
  Copyright (c) 2011 Suat Özgür.  All right reserved.

  Contributors:
  - Andre Koehler / info(at)tomate-online(dot)de
  - Gordeev Andrey Vladimirovich / gordeev(at)openpyro(dot)com
  - Skineffect / http://forum.ardumote.com/viewtopic.php?f=2&t=46
  - Dominik Fischer / dom_fischer(at)web(dot)de
  - Frank Oltmanns / <first name>.<last name>(at)gmail(dot)com
  - Andreas Steinel / A.<lastname>(at)gmail(dot)com
  - Max Horn / max(at)quendi(dot)de
  - Robert ter Vehn / <first name>.<last name>(at)gmail(dot)com
  - Johann Richard / <first name>.<last name>(at)gmail(dot)com
  - Vlad Gheorghe / <first name>.<last name>(at)gmail(dot)com https://github.com/vgheo

  Project home: https://github.com/sui77/rc-switch/

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/

#include "RCSwitch.h"

/* Format for protocol definitions:
 * {pulselength, Sync bit, "0" bit, "1" bit}
 *
 * pulselength: pulse length in microseconds, e.g. 350
 * Sync bit: {1, 31} means 1 high pulse and 31 low pulses
 *     (perceived as a 31*pulselength long pulse, total length of sync bit is
 *     32*pulselength microseconds), i.e:
 *      _
 *     | |_______________________________ (don't count the vertical bars)
 * "0" bit: waveform for a data bit of value "0", {1, 3} means 1 high pulse
 *     and 3 low pulses, total length (1+3)*pulselength, i.e:
 *      _
 *     | |___
 * "1" bit: waveform for a data bit of value "1", e.g. {3,1}:
 *      ___
 *     |   |_
 *
[... 903 lines cut ...]
    int nReceiverInterrupt;
    #endif
    int nTransmitterPin;
    int nRepeatTransmit;

    Protocol protocol;

    #if not defined( RCSwitchDisableReceiving )
    static int nReceiveTolerance;
    static unsigned long nReceivedValue;
    static unsigned int nReceivedBitlength;
    static unsigned int nReceivedDelay;
    static unsigned int nReceivedProtocol;
    const static unsigned int nSeparationLimit;
    /*
     * timings[0] contains sync timing, followed by a number of bits
     */
    static unsigned int timings0[RCSWITCH_MAX_CHANGES];
    static unsigned int timings1[RCSWITCH_MAX_CHANGES];
    #endif

};

#endif
```

fingolfin, 2016-05-23:
Thanks. All in all, I see you did not thing special here, so I am no longer surprised. Rather, you simply replaced the existing hard-coded limitation of a single receiver by another hard-coded limitation (*exactly* two receivers). That is of course not hard to achieve. The difficulties I was alluding to rather are in making the code such that you can simply make one, two or three RCSwitch instances in order to use one, two or three receivers.

Anyway, let me reply to some specific points:

> 1. Sorry, but i have no idea how to do a pull request without forking, so I'm just going to attach my code here

Yes, you need to make a fork in order to create a pull request. That's a feature, not a bug :-). It also makes it possible to properly review your changes, as opposed to just having to look at a big blob of code.

> 2. The code is a proof of concept, not too dirty, but can be much cleaner
> 3. I'm experienced in C/C++ but very new to Arduino

Then you should know that code duplication should be avoided ;-). Your code changes is doing exactly that, by duplicating the interrupt handler code.

> 4. Initially I wanted to go the way of non static, but after reviewing the code, it was much easier to just add another buffer and interrupt handler
> 5. No offence, but I wasn't convinced about the [...]

Sorry, I was unclear in what I wrote: Whenever I wrote `static`, please read `global / static` (note that static has two meanings in C/C++: the one I am referring to is the use of static variables inside a function; the example with `non_static` you give is about the second meaning, which restricts the visibility of a global variable to a file).

So, there is no contradiction here. My point simply was that interrupt handlers can only access global/static variables (and of course anything reachable from that).

Now, it would easily be possible to turn the `static` variables inside the interrupt handler into e.g. globals.

> 6. Assuming my test is correct (aka interrupt handlers can access non static variables), I think the best way is to have 2 instances of rc-switch, 1 for each receiver, keeping as much code and [...]

I perceive two primary problems with your code:
1. It forces a second buffer onto everybody, even people who don't need it (and who need the memory for other things).
2. It duplicates the interrupt handler code.

Problem 1 could e.g. be addressed by a compile time switch, such as an optional `#define USE_SECOND_RECEIVE_HANDLER` at the top of `RCSwitch.h`, to enable/disable compilation of this feature. Or perhaps even better: `#define MAX_NUM_OF_RECEIVERS 2` and then allow setting it to 0 (= disable receiver completly), 1 (= current behavior), 2 (your patch), and possibly even higher.

Problem 2 could be solved by factoring out the state of the interrupt handler into a struct, and the code into a separate function. This then also makes it easy to support more than two receivers, if desired. Here is a rought sketch of how that could look like:
```
struct InterruptHandlerState {
}
```

bilogic, 2016-05-23:
@fingolfin 
You are right, I retested by placing ```non_static``` in the ```RCSwitch.h``` in order to use ```this->non_static``` and it doesn't compile anymore. I know the code is not clean, but it was the quickest way to meet my use case and anyone who needs exactly two receivers. :)

closed 2016-05-23 by bilogic

fingolfin, 2016-05-23:
Of course! Everybody does that, me certainly included :-). Thank you for your input, though, it made me think about this again: While there may be no "perfect" solution, one could still add a working solution which at least helps people who need two receivers; and if done right, it should not "harm" people who need no or just a single receiver.