sui77/rc-switch #70: Guide for Raspberry Pi

98schaeffer, 2016-06-22:
In the code, I see support for the Raspberry Pi.
But how do I implement rc-switch for Raspberry Pi?
I failed to do so.

rotv, 2016-06-22:
Use for instance this project (which in turn uses rc-switch): https://github.com/rotv/433Utils

rotv, 2016-06-22:
Install the wiringPi lib according to the realm and run "make" in the RPi_utils folder to get hold of some useful commands.

98schaeffer, 2016-06-23:
Thank you for the reply. Unfortunately, it won't work. Can you help me? I have installed wiringPi and 433Utils. In RPi_utils I did ´make´ and then ´sudo ./codesend 8853772 1 186´. Nothing happened. I connected everything right I think, data to pin 11 on the RPI.

rotv, 2016-06-23:
I would suspect the pulse length to mismatch. Did you get the receiving command to work?

I've previously used a program called PISCOPE to see allt the signals on all PINs, that's how I could determine the pulse length for my remote.

98schaeffer, 2016-06-23:
I used [rc-switch](https://github.com/sui77/rc-switch) to determine the code, protocol and the pulse length. I managed to control my switch with the arduino using a 433 transmitter. However, I did not succeed to get things work on the RPI.

rotv, 2016-06-23:
Could it be a range thing? I remember having lousy range for my RPi 433 antenna (like a few centimeters) until I soldered a straightened paper clip to it (length 17cm if I recall correctly). Now it's up to 10 meters.

98schaeffer, 2016-06-23:
I have soldered an antenna to it, it works well with the Arduino

rotv, 2016-06-23:
Have you tried receiving through the RPi?

98schaeffer, 2016-06-23:
Using RFSniffer I get 8853772 for ¨on¨. Using ¨sudo ./codesend 8853772 1 186¨ multiple times, the switch went finally on! Perhaps a bad transmitter? Off-topic: the transmitter gets pretty hot. Is that normal?

rotv, 2016-06-23:
Different voltage or current? I occasionally get failed "codesend"s as well bot most often it works. Haven't checked the temperature.

98schaeffer, 2016-06-23:
The transmitter gets less hot when using 3,3V. 
On-topic: I still don´t know why it works with my arduino but not with my RPi

roccomuso, 2016-07-05:
Take also a look at this: https://github.com/roccomuso/iot-433mhz If you want something ready to start working using 433mhz.
It's built based on the rc-switch lib.

Dragao75, 2016-08-11:
When I compile the rc-switch on mu ubuntu 16.04 I get these errors.. 

any one any idea?

make all
g++ -DRPI   -c -o ../rc-switch/RCSwitch.o ../rc-switch/RCSwitch.cpp
g++ -DRPI   -c -o send.o send.cpp
g++ -DRPI  ../rc-switch/RCSwitch.o send.o -o send -lwiringPi
/usr/lib/gcc/arm-linux-gnueabihf/5/../../../../lib/libwiringPi.so: undefined reference to `pthread_join'
/usr/lib/gcc/arm-linux-gnueabihf/5/../../../../lib/libwiringPi.so: undefined reference to `pthread_create'
/usr/lib/gcc/arm-linux-gnueabihf/5/../../../../lib/libwiringPi.so: undefined reference to `pthread_cancel'
collect2: error: ld returned 1 exit status
Makefile:8: recipe for target 'send' failed
make: *** [send] Error 1

SamDecrock, 2017-05-07:
I created some Raspberry Pi examples that are very similar to the Arduino samples: https://github.com/SamDecrock/rc-switch-pi-examples