sui77/rc-switch #75: Problem with make on Raspberry Pi

TimRustige, 2016-07-20:
Current master version download appears to have RCSwitch.o missing? Make on Raspberry PI fails with:

pi@raspberrypi ~/lloytron/433Utils/RPi_utils $ make
make: *** No rule to make target '../rc-switch/RCSwitch.o', needed by 'send'.  Stop.
pi@raspberrypi ~/lloytron/433Utils/RPi_utils $

JerryWorkman, 2016-07-25:
Get the source for RCSwitch here: https://github.com/r10r/rcswitch-pi.git

fingolfin, 2016-07-26:
@TimRustige No, RCSwitch.o is a binary object file, which the C/C++ compiler generates from the source code; as such it should not be in the repository.

Since you are executing a "make" command inside of `433Utils/RPi_utils`, I'd guess that this is a problem (?) with the build system of that project -- or perhaps you missed a step in their build instructions which explains how to compile RCSwitch?

fingolfin, 2016-07-26:
@JerryWorkman Uhm, I don't think it's a good idea to point people to a 4 year old unmaintained fork of RCSwitch.

huongminh, 2016-08-08:
I came across the same issue trying to run "make". then I tried rotv/433Utils and it worked. Thank you!

utsxumiao, 2016-11-05:
@rotv is spot on!

ovidiu-pislaru, 2016-12-03:
go to the root folder of your project. do a `git submodule init` and `git su`

NickSee, 2016-12-09:
thanks  ovidiu-pislaru this also worked for me in terms of getting the make to complete correctly.

jppuertolas, 2018-04-26:
thanks, it also worked for me for building the make