sui77/rc-switch #78: Problem with 32bit and different protocols

WoodsterDK, 2016-08-09:
Hi there,

I have used the library together with an ESP8266, and try to put some codes into a clone remote keyfob.
I would like to use a 32bit value, but cannot get it to receive the learned value. Another thing, if I use standard 24 bit, and transmits a value as a protocol 2 - then the received value gets interpreted as protocol 1. How can that be?
Martin

fingolfin, 2016-08-15:
It's unclear to me from what you write what you tried to do, and what the problem was. Perhaps you can insert some snippets of code you tried to use, and explain what you thought it would do, and what happened instead. Then based on that, perhaps somebody can help you!

1technophile, 2017-04-07:
Hello,

I was trying to implement 32 bit signal on [OpenMQTTGateway](https://github.com/1technophile/OpenMQTTGateway) based on this great library. I encountered problem when trying to work with other length than the traditional 24 that we are seeing on the examples.

Let's see the test I have done.

I modified this code to have some traces:
```
void RCSwitch::send(const char* sCodeWord) {
  // turn the tristate code word into the corresponding bit pattern, then send it
  unsigned long code = 0;
  unsigned int length = 0;
  for (const char* p = sCodeWord; *p; p++) {
    code <<= 1L;
    if (*p != '0')
      code |= 1L;
    length++;
  }
  Serial.println (code);
  Serial.println(length);
  this->send(code, length);
}
```

"test 1"
    mySwitch.send("101000000010101010100"); 
from the trace I see --> code 1312084 length 21 --> KO plug state doesn't change
    mySwitch.send("101000000010101010001"); 
from the trace I see --> code 1312081 length 21 --> KO plug state doesn't change

"test 2"
    mySwitch.send(1312084,21);
--> KO plug state doesn't change
    mySwitch.send(1312081,21);
--> KO plug state doesn't change

"test 3"
    mySwitch.send(1312084,24);
 --> OK plug state change
    mySwitch.send(1312081,24);
 --> OK  plug state change

It seems that when we are not with a length of 24 we are not having the expected behavior. Do you have some infos about this?