sui77/rc-switch #79: SRX882 + STX882 Modules issue

acalatrava, 2016-08-24:
Hello:

I bought SRX882 & STX882 modules and I was able to make the transmitter work with RCSwitch easily. It was just plug&play. Both in Raspberry PI and Arduino.

However I can't make the receiver work. I tested it and it seems that the receiver is working fine. If I use a simple program to attach the interrupt I see the changes when I press a button on my remote. However using the RCSwitch library with the "sniffer" example will not show a thing.

Is there anything I'm doing wrong? or somehow this device is not compatible with RCSwitch? if you could point me in the right direction I will be grateful.

ivandreka, 2016-09-06:
Hello 
I got the same problem as acalatrava has. I guessed it is related to the encoding/decoding scheme. I already ask to the vendor of SRX882 for the similar "RFSniffer" that run in raspberry, but so far there is no answer.
Anybody to share/help is pleased welcome

Martin-Laclaustra, 2016-11-07:
RFSniffer just uses this same library code.
It might be related with the timing responses of the receiver (or even the remote). They might lay off the RCSwitch thresholds. It might differ the timing for up-response and down-response, actually making the up pulses seem fatter or thinner than they should. RCSwitch is rather strict in this sense.
I would recommend to visualize and measure manually the received signal with ReceiveRF.py from [http://www.instructables.com/id/Super-Simple-Raspberry-Pi-433MHz-Home-Automation/?ALLSTEPS]

ivandreka, 2016-11-09:
Hello Martin-Laclaustra. For my project, due to the lack of the SRX882 library for Raspi, I already changed the receiver to other version that have been working fine for Raspi. However, I will give it a try when the project is finished. Thank you for your info.

kolyanok, 2016-11-26:
I have the same problem :(

ivandreka, 2016-11-29:
Hai Martin-Laclaustra, the link you gave is not a SRX882 + STX882 module. So I think it is out-of-topic.
There are some 433Mhz transmitter+receiver that works well with RFSniffer, but nor SRX882.

Martin-Laclaustra, 2016-11-30:
I have no experience with that module, but looking at the specs:
http://g04.s.alicdn.com/kf/HT1FZnrFR0XXXagOFbXy/220307665/HT1FZnrFR0XXXagOFbXy.jpg
([seller](http://wholesaler.alibaba.com/product-detail/ASK-Module-kit-small-size-long_1854611249.html))
it seems that it stays in sleep mode unless pin 4 is at high level. Could it be that there is the need for a pull-up resistor between power and pin 4? (I am no expert, please do this under your own responsibility, and if after reading the specs you arrive to the same conclusion).
An LED between the data pin and ground (or better a connection to the base of a NPN transistor feeding an LED) may help you showing whether the receptor works.
If it does, then the next would be to record and show the signal. The python script in my comment above is a good option. It does not rely on any library. It just plots the signals of the signal pin against time (no protocols involved, so it should not be too many differences between receptors). If you can see a signal, then it is time to check whether RCswitch can decode that signal or it would need some adaptation.
I hope this helps.
[image]

ivandreka, 2016-12-01:
I already set pin no 4 either in high or low state, and both gives the same result -- can not receive 433Mhz signal. It seems that the encoding is different (the width of each pulse width), so in this case your referenced program might be useful for such analysis to find a correct timing scheme. Honestly I do hope that the vendor of the SRX receiver will do this for their customer, as a customer support service,

Frianik, 2017-03-22:
Solved problem connection SRX822 to arduino. 
You have to connect CS pin to vcc arduino pin (sleep off) and change mySwitch.enableReceive(0); to mySwitch.enableReceive(1); 
1 interrupt is pin 2 for 32U4
0 interrupt is pin 2 for 168/328/Mega

Mine Pro micro with 32U4 works well with SRX822

kolyanok, 2017-04-30:
I tried to connect cs pin to 5v on arduino nano, data to D2 pin (mySwitch.enableReceive(0)), and it doesn't work for me :(

jimo3139, 2017-05-22:
I have the same problem. I have CS tied to VCC, and a logic analyzer on the data pin. VCC and GND are hooked up. I have 2 RF remotes, one 433mhz and the other 418mhz. Both work fine with another RF receiver. RXB4411S-418-RH

mtgithubmt, 2017-11-09:
Hi,
I tried all of the above and it did not work however I managed to get it working.
I changed the baud rate to 115200. It even worked without the antenna. Here is my code for the receiver.
/*
  Simple example for receiving
  
  https://github.com/sui77/rc-switch/
*/

#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(115200);
  mySwitch.enableReceive(0);  // Receiver on interrupt 0 => that is pin #2
  Serial.println("start");
}

void loop() {
  if (mySwitch.available()) {
    
    int value = mySwitch.getReceivedValue();
    
    if (value == 0) {
      Serial.print("Unknown encoding");
    } else {
      Serial.print("Received ");
      Serial.print( mySwitch.getReceivedValue() );
      Serial.print(" / ");
      Serial.print( mySwitch.getReceivedBitlength() );
      Serial.print("bit ");
      Serial.print("Protocol: ");
      Serial.println( mySwitch.getReceivedProtocol() );
    }

    mySwitch.resetAvailable();
  }
}

Wiring from right to left. 
(GRD, PIN2, EMPTY (not connected to 5v), 5v )
Wiring for antenna. (Right to left)
(GRND (bottom 2nd from left hole), ANTENNA (bottom left hole), ANTENNA (top left hole))

Hope this helps.

mtgithubmt, 2017-11-09:
PS I am receiving the information from my PIR 433 transmitters from my alarm.

jeffersonfpalheta, 2017-11-14:
A strange thing is happening... a have a professional Oscilloscope and I was analysing the Data signal in the SRX882 V1.3 module then I noticed that the level of the signal sent falls down from 5v to 1.2v approximately. I made another test with some resistors and it happens in the same way. So I'm sure it shouldn't happen then I think something is wrong, I even tested another modules of the same model but nothing. So it doesn't make sense for me. Could someone give me a light?

runesoeknudsen, 2018-01-06:
I having similar issues. It works when the SRX882 data pin connected to my oscilloscope. But then I connected it to my ESP8266, GPIO4, the signal disappears.
What might explain it is that the datasheet for the IC on the SRX882 module, PT4303,  says that you only can expect the voltage level, to be between 0.1 to 0.9V at 1µA. And that it at max only can drive 20µA. Please correct me if I am wrong!
[image]

see [http://www.dzsc.com/uploadfile/company/307703/201246135150265.pdf](http://www.dzsc.com/uploadfile/company/307703/201246135150265.pdf), page 5 at the bottom

AlexaAW, 2018-03-21:
This receiver model has a high output impedance at the "date" output. For correct operation, the microcontroller input must also have a high input impedance. For this, the programs need to disable internal pull-up resistors. And eliminate external load circuits with a resistance below 10 kΩ

MACscr, 2018-09-20:
I am still lost on this. Cant for the life of me get the SRX882 to work with rc-switch with even the basic receivedemo with my esp8266-07. Ive tried both d2 and d3 and changed the code from mySwitch.enableReceive(0) and mySwitch.enableReceive(4); according to what i am seeing in my breakout https://www.screencast.com/t/0ffXXG7MtgqL. Do we need to do something with the code or is it a physical change that has to be made to get it to function? Amazing as this seems to be a well promoted receiver, but very little information on actually getting it to work.

jeanniquini, 2018-11-16:
It worked on the D1 pin on my nodemcu.

opodul, 2018-12-19:
Hi Guys,
I got the same issue and google brought me here.
Your discussion put me on the correct track and now the problem is solved.
Indeed, it comes from high impedance output from the receiver witch do not provide enough current to the pull-down resistor of the raspberry Pi (enable by default).

The solution is to call : pullUpDnControl( PINRX , PUD_OFF );    (from wiringPi lib)
which disable the pull_Up/Down resistors. On the oscilloscope I have now a nice 0 to 3.3V signal again.

Just for fun I also played around with  :
pullUpDnControl( PINRX , PUD_DOWN ); the result is a signal from 0 to 1.1V
pullUpDnControl( PINRX , PUD_UP ); the result is a signal from 2.2V to 3.3V

Other topic: To limit the number of false data (unwanted pulse due to noise) I put capacitor of 220nF between Data and Gnd. It would be nice to know how to limit the sensitivity of this PT4303…

AlexAAW, 2018-12-20:
Putting a condenser on the data line is a bad idea. Especially 220n. With such a high resistance of the output stage, this capacitance will fill up the fronts so much that during the pulse repetition time it will not reach a high level, and errors will only increase.
The best way to deal with interference is software.
It is necessary to teach the system to respond to pulses of a certain duration (it will be something like a digital filter)

sebstardoo, 2019-02-05:
Hey guys, I feel like this forum didn't really help me for an arduino setup so now that I've managed to get it working i'd share my code and setup:

pin connections
TX side:
VCC - 5v
GND - GND
DATA - pin d7 (arbitrary, just had pin 7 free at the time but any will do)

RX side:
VCC - 5v
GND - GND
CS - 5v
DATA - pin d5

TX code:

#include <RH_ASK.h>
#include <SPI.h> // Not actually used but needed to compile

RH_ASK driver(2000, 7, 9, 10); // speed,rx,tx,mode

void setup()
{
    Serial.begin(9600);	  // Debugging only
    if (!driver.init())
         Serial.println("init failed");
}

void loop()
{
    const char *msg = "hello, this is a test";

    driver.send((uint8_t *)msg, strlen(msg));
    driver.waitPacketSent();
    digitalWrite(13, HIGH);
    //this last part is just to toggle the onboard led after each transmission
    delay(50);
    digitalWrite(13, LOW);
    delay(200);
}

RX side:

#include <RH_ASK.h>
#include <SPI.h> // Not actualy used but needed to compile

RH_ASK driver(2000, 5, 2, 10); // speed, rx, tx, mode

void setup()
{
    Serial.begin(9600);	// Debugging only
    if (!driver.init())
         Serial.println("init failed");
    Serial.println("begin");
}

void loop()
{
    uint8_t buf[RH_ASK_MAX_MESSAGE_LEN];
    uint8_t buflen = sizeof(buf);

    if (driver.recv(buf, &buflen)) // Non-blocking
    {
	int i;

  String str = "";

  for (int i=0; i < buflen; i++) {
    str += (char)buf[i];
  }

  Serial.println(str);
    }
}

Final note, this was done on arduino nano clones, both atmega 328 and 168 but i can't see why it wouldn't work on any other arduino platform

Georgex12, 2019-02-07:
Ok this thread got very messy.....here is how I got it to work.

1. You have the **receiver SRX882**
2. You would like to use rs-switch and **sniff the code of your remote control outlet's**...
3. You have the **SRX882 connected to an arduino nano** and it's not working with rc-switch.....

Open this repository on GitHub: 
https://github.com/RoyOltmans/OpenMQTTgatewayLight/blob/master/lib/NewRemoteSwitch/examples/ShowReceivedCode/ShowReceivedCode.ino

You will figure out how to download and integrate all necessary files into your arduino library.... You just need the folder : **NewRemoteSwitch** / Open example "**ShowReceivedCode**".

Wireing: 
Connect GND to GND, VDD to arduino nano 5V, DATA to arduino nano D2, CS to second 5V pin on arduino nano. **CS must be connected to high** (have 5V) ! The arduino is connected to serial on your PC (USB)...so open serial monitor of your arduino IDE. After a moment you will see : "Receiver initialized". Go ahead and press button on your remote control...TADA !!! You see code & period....

jnovos, 2023-12-20:
Hello,
For what it's worth, the reception module works with **external interrupts**. I have a nano AtMega328p and it only works by connecting it to pin D2 or D3 and the CS at 5V.
```c++
#include <RCSwitch.h>
RCSwitch mySwitch = RCSwitch();
void setup() {
  Serial.begin(9600);
  delay(1000);
  Serial.println("Press 'Send' to start programming or press a key on remote to see the value.");
  mySwitch.enableReceive(0);  // RF Receiver D2 o D3 mySwitch.enableReceive(1);  
}
void loop() {
  while (!Serial.available()) {    // Receive Key Codes until Send was pressed
    if (mySwitch.available()) {
      long unsigned int value = mySwitch.getReceivedValue();
      Serial.print("Received Key: ");
      Serial.println(value);
      mySwitch.resetAvailable();
    }
  }
  while (Serial.available()) Serial.read();
  Serial.println();
}
```

ivandreka, 2023-12-21:
hi , thank you for your reply
excellent 👍

best regards

Zhan85, 2024-01-03:
I have an srx822s.  Tried the above 2 codes.  Putting 5v on CS and VCC,  GND to GND, and tried data on pin 2 and then pin 3 of an Arduino Uno.  Cannot receive any code from my RC300 fireplace remote which operates at 433.91mhz.  The prompts both come up on the serial monitor so I know the code is working.  Is there anything I am forgetting or can check?