sui77/rc-switch #80: signal not received

bilogic, 2016-08-25:
Hi,

I have tried using both arduino uno and esp8266 to receive 433Mhz signals from those power socket transmitters, but they are not as responsive as the power sockets. Meaning, I need to hold down my transmitter button a little longer for uno or esp8266 to receive and echo on screen.

Is there any possible room for improvement? Thank you.

fingolfin, 2016-09-06:
Based on what little information you provided: Sorry, can't help you. It is completly unclear what your setup is, what you exact problem is, what those "power sockets" you talk about are, etc.

Also, this is not really a support forum, but rather a tracker of submitting bug reports with rc-switch (although we *do* try to help with problems, but only if they are very clearly described).

I recommend reading http://www.catb.org/esr/faqs/smart-questions.html and then perhaps trying again, with much more detail, e.g. on http://arduino.stackexchange.com/

closed 2016-09-06 by fingolfin

bilogic, 2016-09-06:
Hi, the reason I'm asking here is because the solution probably involves modifying the codes.
Hope you can give me the chance to describe my problem 1 more time.

1. I have a set of 433 Mhz TX and RX and using your library, I came to know they use 32bit codes
1. The TX is a simple button remote control, while the RX is a on/off switch
1. I can toggle the RX on/off using the TX by pressing the button
1. I also have a working circuit using your library and uno to receive this 32bit 433 Mhz signal.
1. The circuit will listen for the signal from the TX and display "Received" via uno's serial output
1. The circuit displays "Received" when I press the button on the TX to send the 32bit 433 Mhz signal.
1. My problem is that for the circuit to display "Received", I need to hold down the TX button a little longer than in step 3
1. My question is, can this library be made to react faster? (i.e. on par with the RX reaction speed)
1. I did try using the esp8266, which runs at a higher clock speed, but it did not help. I still have to hold down the TX button longer for the circuit to register the signal.
1. I'm no RF expert, but I saw some variables in the code which led me to wonder if they can be adjusted.

Thank you for your time.

silas2222, 2020-03-26:
still no fix 2k20 wtf

SymbolCsaba, 2020-08-12:
This is an abbandoned repo, but base of some super projects! Thanks for sui77!
I found a fork, that can handle more receiver in one time (problem was the common interrupt handler). I have made one imporevement to reduce receiving time. See this repo and examples: https://github.com/SymbolCsaba/RCSwitch

svicar, 2022-01-11:
BCsabaEngine...Why I cannot ude your code on nano? On ESP32 works faster than original.

BCsabaEngine, 2022-01-11:
On ESP32 works faster than original: this is the reason why setReveiceRepeat(0) is important.

Cannot use your code on nano: what is the problem, compile or usage errors occurs?

svicar, 2022-01-11:
there is no issues on  https://github.com/BCsabaEngine/RCSwitch
How to write there?

BCsabaEngine, 2022-01-11:
> there is no issues on https://github.com/BCsabaEngine/RCSwitch How to write there?

Open one with your issue.

svicar, 2022-01-11:
[image]
Issues is missing...where to write?

BCsabaEngine, 2022-01-12:
> [image] Issues is missing...where to write?

Changed, issues available. Sorry