sui77/rc-switch #84: Random crash when interrupt is called.

gadget1999, 2016-09-06:
This is related to issue #46 , it seems I need to add ICACHE_RAM_ATTR to RCSwitch::handleInterrupt() to solve it.

Version used: rc-switch 2.5.2
Hardware: NodeMCU v2 (ESP-12E)
Arduino ver: 1.16.10

I also tried the following macro, compiler reported "#error  ESP8266 is defined"
#ifdef ESP8266
#error  ESP8266 is defined
#else
#error ESP8266 is NOT defined
#endif

Please see if something I missed or it's a bug? Thanks!

anil8anil, 2019-04-03:
Can you share which lines did you changed in cpp files and how? I am in a similar situation... Thanks

gadget1999, 2019-04-03:
I changed RCSwitch.h
    static void ICACHE_RAM_ATTR handleInterrupt();

It's quite long ago, so the line number may not match up with current code.

anil8anil, 2019-04-07:
Unfortunatelly i couldnt fix my problem...