sui77/rc-switch #88: Superswitch, can receive and decode, but can't send

badabing2005, 2016-10-03:
Hi,

Thanks for the awesome library and great documentation.
Which helped me capture and decode SuperSwitch RF plugs
[image]

Button 1
```
Decimal: 7632402 (24Bit)
Binary: 011101000111011000010010
Tri-State: not applicable
PulseLength: 226 microseconds
Protocol: 1
Raw data: 19868,20,232,100,16,28,88,32,20,116,20,288,28,656,52,108,136,188,132,184,36,48,16,44,120,52,20,876,20,484,12,136,20,328,1932,16,32,20,228,112,16,704,32,196,192,12,100,320,20,
```

Button 2
```
Decimal: 7632404 (24Bit)
Binary: 011101000111011000010100
Tri-State: not applicable
PulseLength: 226 microseconds
Protocol: 1
Raw data: 7024,256,396,44,168,152,80,12,24,20,592,484,36,260,64,216,52,100,44,660,24,364,56,192,24,16,16,272,56,444,60,144,48,324,56,172,68,624,148,248,28,856,44,756,44,172,100,3088,28,
```

Button 3
```
Decimal: 7632408 (24Bit)
Binary: 011101000111011000011000
Tri-State: not applicable
PulseLength: 226 microseconds
Protocol: 1
Raw data: 7028,240,668,716,208,696,216,704,216,344,16,1704,348,108,92,132,216,100,60,12,132,68,28,248,92,148,68,104,32,260,20,996,68,40,100,92,224,88,68,32,28,96,96,136,244,96,44,308,56,
```

Most likely I'm missing something simple rather than this being a bug somewhere, but for the life of me I'm not able to turn on/off the plugs using the transmitter.

I noticed that no matter what value I set for the setPulseLength, I always get a PulseLength of 353 microseconds, which seems weird

Let me explain.
My receiver is this
https://www.digikey.ca/product-detail/en/rf-solutions/QAM-RX2-433/QAM-RX2-433-ND/2000715
My first transmitter is this
https://www.digikey.ca/product-detail/en/rf-solutions/QAM-TX2-433/QAM-TX2-433-ND/2781776
My Second transmitter is this
https://shop.openenergymonitor.com/ook-on-off-keying-rf-transmitter-433mhz/
Which practically is the same as the first one.

The SuperSwitch is using EV 1527

At first I just used the transmitter, with a very simple sketch

The sockets wouldn't turn on/off
Then I adapted the sketch to see if it was transmitting properly, and captured it through the receiver
See two transmitters (pin 9 and 10 on Arduino) and 1 receiver.
[image]

Here's the modified sketch to send and receive (I'm sending button 1 and button3 commands)
```
/*
  Example for receiving
  
  https://github.com/sui77/rc-switch/
  
  If you want to visualize a telegram copy the raw data and 
  paste it into http://test.sui.li/oszi/
*/

#include <RCSwitch.h>

RCSwitch mySwitchRx = RCSwitch();
RCSwitch mySwitchTx = RCSwitch();
RCSwitch mySwitchTx2 = RCSwitch();

void setup() {
  Serial.begin(9600);
  mySwitchRx.enableReceive(0);  // Receiver on interrupt 0 => that is pin #2

  // Transmitter is connected to Arduino Pin #10  
  mySwitchTx.enableTransmit(10);
  mySwitchTx2.enableTransmit(9);

  // Optional set pulse length.
  mySwitchTx.setPulseLength(840);
  mySwitchTx2.setPulseLength(226);
  
  // Optional set protocol (default is 1, will work for most outlets)
  mySwitchTx.setProtocol(1);
  mySwitchTx2.setProtocol(1);
  
  // Optional set number of transmission repetitions.
  // mySwitchTx.setRepeatTransmit(15);

  pinMode(13,OUTPUT);

}

void loop() {
  if (mySwitchRx.available()) {
    output(mySwitchRx.getReceivedValue(), mySwitchRx.getReceivedBitlength(), mySwitchRx.getReceivedDelay(), mySwitchRx.getReceivedRawdata(),mySwitchRx.getReceivedProtocol());
    mySwitchRx.resetAvailable();
  }

  mySwitchTx.send("111100100100110011111000");
  digitalWrite(13,HIGH);
  delay(200);
  digitalWrite(13,LOW);

  delay(2000); 

  mySwitchTx2.send("011101000111011000010010");
  digitalWrite(13,HIGH);
  delay(200);
  digitalWrite(13,LOW);

  delay(2000); 
}
```

And here's what gets captured in the serial monitor
```
Decimal: 15879416 (24Bit) Binary: 111100100100110011111000 Tri-State: not applicable PulseLength: 353 microseconds Protocol: 1
Raw data: 7960,1224,1700,24,136,596,256,1868,132,724,392,108,280,244,852,752,24,188,624,716,1716,168,252,160,164,176,52,116,172,48,184,296,476,140,168,144,2684,108,2260,136,472,276,96,184,548,84,124,104,92,

Decimal: 7632402 (24Bit) Binary: 011101000111011000010010 Tri-State: not applicable PulseLength: 353 microseconds Protocol: 1
Raw data: 6140,2404,512,16,1940,1180,224,40,540,80,1788,32,936,36,132,16,3416,36,172,344,96,492,164,28,208,360,168,32,172,24,176,32,76,168,96,156,116,68,200,36,232,36,452,40,188,272,868,24,2484,

```

Notice how both capture Pulse Lengths are 353 microseconds, even though I set them to 840 and 226
(226 was the value captured from the original switch)

Can anyone spot anything wrong I'm doing?
I know I don't have an Antenna, but I was able to capture without one, and I'm told, it's supposed to work for few meters without an Antenna, and yet it does not work right next to the socket.

PS: I tried 17cm wires straight and coiled (see image) when receiving / transmitting
[image]

I got much better results using a 19.5 cm connector wire (when receiving)
[image]

Thanks in advance

badabing2005, 2016-10-23:
Thanks YFEDOROV624,
That explains it, I was wondering why all of a sudden it was working in my latest sketch.
Because I have setPulseLength() after setProcotol().

Thanks for the clarification.
I also realized that my power sockets worked on 315 Mhz, hence why the antenna problems.

Now everything is working smoothly.
Many thanks

closed 2016-10-23 by badabing2005

novahawk13, 2016-10-28:
I just ran into this exact issue.  The Send Demo provided with the library has the two lines reversed which is where the confusion is coming from.

`  // Optional set pulse length.
  // mySwitch.setPulseLength(320);
  
  // Optional set protocol (default is 1, will work for most outlets)
  // mySwitch.setProtocol(2);`

Thanks @YFEDOROV624

quixand, 2018-07-07:
there really needs to be some documentation about setPulseLength and setProtocol or a wiki entry. they seem like really important functions