sui77/rc-switch #92: ReceiveDemo_Advanced not working on Arduino UNO

virkon, 2016-11-13:
Hello there. Thanx a lot for your excellent work. I've been fairly new to this and I am probably asking a dumb question, but I read all closed issues and googled around and still I am unable to solve this issue. 
I am using 2 Arduino UNO and the simple MX-05V and MX-FS-03V transmitter/receiver modules. Using a simple example based on the radiohead library (http://www.airspayce.com/mikem/arduino/RadioHead/RadioHead-1.41.zip) I can confirm, the TR/RC Modules are indeed working. I did the Hello World example here: http://randomnerdtutorials.com/rf-433mhz-transmitter-receiver-module-with-arduino/

Knowing this I set up the RC Module with an Arduino UNO and the Data Pin on Interrupt 0 (Pin2) and tried the ReceiveDemeo_Advanced Example. But nothing on the Serial Monitor. By attaching a LED between GND and the Data Pin I can literally see that the transmit is physically working, but for some reason the Arduino is not displaying anything on the Serial Monitor. 

What am I missing?

CrosseyeJack, 2016-11-15:
IIRC ReceiveDemeo_Advanced will only output when it gets data its able to decode the protocol for it so manybe the switch your trying to read isn't using a supported protocol.

virkon, 2016-12-03:
CrosseyeJack, thank you. I thought as much, but since I am a total n00b to this I wanted to confirn. Have to learn more about decoding those signal then. Thanx for the feedback.
regards, v.

closed 2020-01-05 by 1technophile