sui77/rc-switch #96: Problem recieving anything

Edvin85, 2016-11-27:
Hi, looks like a great library, but i have some problems getting this to work.
I uploaded RecieveDemo-simple example to my UNO R3 board, (also tried on MEGA2560) using Adruino version 1.6.13.
Under upload i get this warning in the window:
"WARNING: Category 'Device Control, Signal Input/Output' in library rc-switch is not valid. Setting to 'Uncategorized'"
I connected reciever as described with the data pin to interrupt 0 (D2).
When looking in the serial monitor with baud9600, nothing happens
I have beed trying those units:
Nexa handheld switch
Nexa Wall switch
Nexa timer remote
ESIC temperature transmitter

I have been using this reciever to other home automation projects with other librarys with success, an i can see the reciever picup signals from all units if i connect a led to the datapin.

What am i missing here?

Best regards

alexschwantes, 2017-01-02:
I could be wrong, but I don't think nexa is supported..
At least I couldn't get it to work.

bphermansson, 2017-01-09:
Just tried with a Arduino Duemilanove and Nexas PAT-103 remote, works fine.

Totoche95, 2017-03-06:
I also connected a receiver on D2 (i see signal on scope) and program an Arduino Uno with simple_receive sketch. Nothing on serial.
Is there a line to insert in rc-switch.cpp or .h to see if interrupt is activated and have a message even the protocol is different ? I program in C but nothing in CPP.

Totoche95, 2017-03-06:
OK, i commented the line 
// detect overflow      
if (changeCount >= RCSWITCH_MAX_CHANGES) {

So it accept long frames, it cannot decode it (it is a 433MHz thermometer sensor) but i know it works fine.
Cheers

Received 484 / 4032bit Protocol: 1960
Unknown encodingReceived 4032 / 488bit Protocol: 484
Received 4032 / 484bit Protocol: 488
Received 4028 / 484bit Protocol: 488
Received 4028 / 488bit Protocol: 492
Received 1952 / 488bit Protocol: 488
Received 4044 / 468bit Protocol: 468
Received 4048 / 468bit Protocol: 472

alexschwantes, 2017-03-30:
Just to further clarify why it didn't work for me. 
I had tried to get it working with a Nexa PE-3 remote (https://www.nexa.se/vara-produkter/system-nexa/paket/pe3) and it didn't work as it uses a different (newer) protocol. These blogs describe the newer protocol http://tech.jolowe.se/home-automation-rf-protocols/ , http://dephiox.blogspot.no/2013/01/decoding-new-nexa-protocol.html 

I used https://github.com/JoakimWesslen/Tx433_Nexa as a base along with piscope to find the code being sent. It worked after a lot of effort, but in the end I found RF just not as reliable as a wifi switch.