sui77/rc-switch #97: mySwitch.switchOff causes Exception 28

comdata, 2016-11-29:
Hello,

when calling the switchOff multiple times in a row it causes and Exception 28.

This is the trace from the serial console:

Exception (28):
epc1=0x4020a962 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffefd00 end: 3ffeffe0 offset: 01a0

>>>stack>>>
3ffefea0:  3ffeea00 3ffeeafc 3ffeeefc 4020bf18  
3ffefeb0:  0000003f 3ffeeafc 3ffeeefc 40206e4b  
3ffefec0:  74617473 30007375 756f6800 6f436573  
3ffefed0:  31006564 31303031 646f4300 203a2265  
3ffefee0:  30313122 2c223130 7322090a 63746977  
3ffefef0:  226f4e68 3122203a 30303030 407d0a22  
3ffeff00:  00000000 00000000 00000000 4020af0a  
3ffeff10:  00000000 00000000 00000000 00000002  
3ffeff20:  3ffeea1f 0000003e 3ffeea5d 4020b0e5  
3ffeff30:  407d004a 00001388 00001388 3ffeff70  
3ffeff40:  3ffeff70 00014dd4 3ffeea10 4020dc5c  
3ffeff50:  3fffdad0 00014b8f 3ffeea10 4020b672  
3ffeff60:  6977732f 00686374 3ffeea10 4020b588  
3ffeff70:  3ffe8901 3ffeeefc 3ffeea10 4020758d  
3ffeff80:  00000000 00000000 3ffeefc0 00000000  
3ffeff90:  3fffdad0 3ffeefb8 4020cce8 3ffeefb8  
3ffeffa0:  3fffdad0 00000000 3ffeea10 402075b8  
3ffeffb0:  00000000 00000000 3ffeefb1 402075d3  
3ffeffc0:  3fffdad0 00000000 3ffeefb1 4020cd34  
3ffeffd0:  feefeffe feefeffe 3ffeefc0 40100718  
<<<stack<<<

What could be wrong?

Kind Regards,
Christoph

sui77, 2016-12-22:
https://github.com/esp8266/Arduino/issues/1275
is it a esp?

comdata, 2016-12-22:
Yes it is a NodeMCU with an ESP 12E.

sui77, 2016-12-22:
i'll try to reproduce that at times

comdata, 2016-12-22:
If you need anything or I should test some things I am more than willing to help.

sui77, 2016-12-22:
this works fine with my esp12e. can you provide your code?

```
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(9600);
  mySwitch.enableTransmit(10);
}

void loop() {
  mySwitch.switchOff("11111", "00010");
  //delay(100);
  //Serial.println("something");
}
```

comdata, 2016-12-26:
Please find my code below. I am using MQTT to trigger the signal.

`

// includes
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <FS.h>
#include <ArduinoOTA.h>
#include <RCSwitch.h>
#include <PubSubClient.h>
#include <ArduinoJson.h>

StaticJsonBuffer<200> jsonBuffer;

 char* status;
    char* houseCode;
    char* switchNo;

/**
   @brief mDNS and OTA Constants
   @{
*/
#define HOSTNAME "ESP8266-OTA-" ///< Hostename. The setup function adds the Chip ID at the end.
/// @}

/**
   @brief Default WiFi connection information.
   @{
*/
const char* ap_default_ssid = ""; ///< Default SSID.
const char* ap_default_psk = ""; ///< Default PSK.
/// @}

RCSwitch mySwitch = RCSwitch();

const char* mqtt_server =  "iot.eclipse.org"; //"192.168.1.76";

WiFiClient espClient;
PubSubClient client(espClient);

/// Uncomment the next line for verbose output over UART.
//#define SERIAL_VERBOSE

/**
   @brief Read WiFi connection information from file system.
   @param ssid String pointer for storing SSID.
   @param pass String pointer for storing PSK.
   @return True or False.

   The config file have to containt the WiFi SSID in the first line
   and the WiFi PSK in the second line.
   Line seperator can be \r\n (CR LF) \r or \n.
*/
bool loadConfig(String *ssid, String *pass)
{
  // open file for reading.
  File configFile = SPIFFS.open("/cl_conf.txt", "r");
  if (!configFile)
  {
    Serial.println("Failed to open cl_conf.txt.");

    return false;
  }

  // Read content from config file.
  String content = configFile.readString();
  configFile.close();

[... 280 lines cut ...]
    // Create a random client ID
    String clientId = "ESP8266Client-";
    clientId += String(random(0xffff), HEX);
    // Attempt to connect
    if (client.connect(clientId.c_str())) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      client.publish("/status", "alive");
      // ... and resubscribe
      client.subscribe("/switch");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}
`

This fails after some loops.

Kind Regards,
Christoph

comdata, 2016-12-31:
I found the issue.

The declaration of `StaticJsonBuffer<200> jsonBuffer;` at the top was causing problems. I moved it into the callback function and now it is working.

Thanks
Christoph

closed 2016-12-31 by comdata