sui77/rc-switch #98: Send signal not working

gerhick, 2016-11-30:
Hi all, 

I'm trying to set up a 433mhz transmitter on my Freetronics Leostick (similar to leonardo), but it's simply not working, and I can't be more specific than that...

Here's what I've tried:

I've used the same 433mhz transmitter with my Raspberry Pi to test it, and it's working fine - also sending the same signal.

I've used the receive code that comes with the RCSwitch library, and it works fine.

I added LED Blink code to the sketch to indicate that something should be happening.

The sketch I'm using is below, and I've also attached an image of the setup
[image]

`/*
  Example for different sending methods
    https://github.com/sui77/rc-switch/
  */

#include <RCSwitch.h>
RCSwitch mySwitch = RCSwitch();
void setup() {
  Serial.begin(9600);
  
  // Transmitter is connected to Arduino Pin #10  
  mySwitch.enableTransmit(10);

  // Optional set pulse length.
  // mySwitch.setPulseLength(320);
  
  // Optional set protocol (default is 1, will work for most outlets)
  // mySwitch.setProtocol(2);
  
  // Optional set number of transmission repetitions.
  // mySwitch.setRepeatTransmit(15);
    pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  /* Same switch as above, but using decimal code */
  mySwitch.send(8442124, 24);
  delay(1000);  
  mySwitch.send(8442116, 24);
  delay(1000);  

  /* Same switch as above, but using binary code */
  mySwitch.send("100000001101000100001100");
  delay(1000);  
  mySwitch.send("100000001101000100000100");
  delay(1000);

  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);  
}`

haoudoin, 2016-12-16:
@gerhick Hope this answer gets to you in time. Try setting the Pulse Length.  189 worked for me with Etekcity RF plugs.

```
/ Optional set pulse length.
// mySwitch.setPulseLength(320);
mySwitch.setPulseLength(189);
```

Guess that line isn't so, "optional"... :)

Hope that helps!

baicaigithub, 2018-12-22:
I tried to figure out the best length for my etekcity rf switch.
The working frequency is 105 - 210. So I chose the middle - 105. I tried with 20 times without a failure.

1technophile, 2020-01-05:
Thanks for the informations. As you found the working value I close the issue.

closed 2020-01-05 by 1technophile