sui77/rc-switch #105: Transmitting and receiving other values

Castrovalva, 2017-01-16:
I know this library is not meant to send values. But is it possible to do something like this:

`mySwitch.send("Hello World!");`

And then receive it:

```
void loop() {
  if (mySwitch.available()) {

    int value = mySwitch.getReceivedValue();

    if (value == 0) {
      Serial.print("Unknown encoding");
    } else {
      Serial.print("Received ");
      Serial.print( mySwitch.getReceivedValue() );
      Serial.print(" / ");
      Serial.print( mySwitch.getReceivedBitlength() );
      Serial.print("bit ");
      Serial.print("Protocol: ");
      Serial.println( mySwitch.getReceivedProtocol() );
    }

    mySwitch.resetAvailable();
  }
}
```

I have tried this, but this won't work, probably because mySwitch.available() only looks for specific data.

sui77, 2017-01-18:
You could send every character one by one.

String text = "Hello World!";
for(int i=0; i<11; i++) {
   mySwitch.sendDecimal( (int)text[i] );
}

And convert it to a character back with
 Serial.print( (char)mySwitch.getReceivedValue() );

but...

The 433 OOK transmission isn't really robust enough to be used to send real messages. Every code is sent multiple times and eventually correctly detected once and there is no error correction. That's sufficient to send single on/off commands but I'd recommend Bluetooth, Wifi or RFM12 for your needs.

closed 2017-02-07 by fingolfin

mrmattwilkins, 2018-02-26:
You can send strings, it just takes a bit of work.

Send a string, one byte at a time.

RCSwitch repeats transmissions, so the receiver will get each character an
undetermined amount of times.  So when sending HELLO, the receiver may get
HHHHEELLLLLLLLOOO and it is impossible to know if the message is HELO, HHELLO,
etc etc.

To get round this problem prefix each byte with a number starting from 0, then
incremented for each character.  So HELLO is sent as 0H1E2L3L40.  The
receiver may get 0H0H0H1E1E2L2L2L2L3L3L40, but can easily work out the
message

Note that you should append an end-to-end checksum, because it is possible that
a character can be completely dropped.

Also note that I couldn't send a 0, but the end of a string can be denoted by 255

void send_string(const char *str)
{
   int i = 0;

   for (char *p = str; *p; p++ ) {
      mySwitch.send((i<<8) + *p, 16);
      i++;
   }
   mySwitch.send((i<<8) + 255, 16);

}

On the receiving end just ignore repeated messages, so if you get
  0H 0H 0H 1E 1E 2L 2L 3L 3L 3L 40 40 5<255> 5<255>
where <255> is just byte 255 but I couldn't display it, just drop all the repeats to get
  0H 1E 2L 3L 4O 5<255>
then remove the leading number byte