sui77/rc-switch #106: FHT-E-WS-7901-WH support

vshymanskyy, 2017-01-18:
I have these FHT-E-WS-7901-WH, sold in Ukraine.
I did some investigation... it's basically same FHT7901, but has a bit different coding scheme.
It has 10 DIP switch, but encodes group settings to tristate like this:
1->1
0->F
The device is encoded as usual:
1->0
0->F

I wonder if we could add support for this type of coding out of the box? Thx!

sui77, 2017-01-18:
you could use the sendTriState("F10...") method right away. 

but i'm not sure how to best name/implement a convenience switchOn/Off() method

vshymanskyy, 2017-01-18:
Sure, it works well with ```sendTriState```.
Yes, naming a convenient method is a pain ;) Maybe in can be implemented as a separate function,
then wait for more such cases to appear and to re-factor into something meaningful.