sui77/rc-switch #111: IR Remote & RC Switch libraries incompatibility

BlackyS, 2017-02-02:
Hello,
In a small project, I would like to receive an IR Signal from a TV Remote (=> I'm using the IRremote library https://github.com/z3t0/Arduino-IRremote here) - and trigger a 433MHz switch in order to switch it on/off.

Both library are working fine separately (!!!), IR remote as receiver and RC switch as transmitter, but when **use those together** it doesn't work and the RC switch library doesn't send anything to my light switches devices (but I'm still able to receive/decode IR signals)

I found some references where others are facing a similar problem and the suggestion was, that maybe there is a timer/interrupt issue which is disturbing those to libraries.
(IRremote is using 1x timer + interrupt ... and a tried it with both timer1 or timer 2 of my ATmega328/Nano ... = same behaviour) 

But what is confusing to me: in the RC Switch send() function I could not identify any interrupt or timers, hence how could this interfere than with IRremote?

Another strange thing is: my piece of code was in a working environment for more than a 1/2 year now, and I just wanted to replace my broken RC-433MHz-Plug with a new one ... so I just wanted to changed the 433MHz Datagram (to match the new Plug) and after the new compiling/upload it stopped working....
... very confusing ...

Any idea what is going on here?
THX!!!

BlackyS, 2017-02-03:
I guess I found a - maybe not very clean - solution ... but better than nothing.

I now simply disabled all interrupt before calling the RC-funtion send() -in order to send a 433MHZ datagram, and enable all interrrupts right after it.

```
  noInterrupts();
        myRCSwitch.send(17492,24);
  interrupts();
```
... as I said: better than nothing :)

1technophile, 2017-04-07:
Hello,

Take a look at the software part of this tutorial:
https://github.com/1technophile/OpenMQTTGateway/wiki/Setup-Arduino

I have both libraries working well on arduino after the having modified IRremote/boarddefs.h as explained.