sui77/rc-switch #117: setPulseLength doesn't work, the PulseLength won't change

quaec, 2017-02-10:
Hello,

I'm trying to control a 433MHz outlet with this library, but it seems that setting the pulseLength doesn't work.

I'm using an Arduino UNO R3 and a FS1000A sending module.

The original remote has a PulseLength of 144 microseconds, but my sender sends always a signal with a PulseLength of 352 microseconds, even if I change it in the sketch.

I've captured the **output of the original remote**:

```
Decimal: 6247424 (24Bit) Binary: 010111110101010000000000 Tri-State: FF11FFF00000 PulseLength: 144 microseconds Protocol: 1
Raw data: 4536,152,428,424,160,120,464,420,160,420,172,408,176,408,176,404,176,108,476,404,180,104,480,400,184,100,480,400,184,100,484,396,184,104,480,400,184,100,480,400,184,396,188,396,188,396,184,396,188,
```

The **sketch** (based on the SendDemo): 

```Arduino
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
  // Transmitter is connected to Arduino Pin #10
  mySwitch.enableTransmit(10);
  mySwitch.setPulseLength(144);
  mySwitch.setProtocol(1);
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);       // turn the LED on (HIGH is the voltage level)
  mySwitch.send("010111110101010000000000");   // Switch ON
  
  delay(500); 
  digitalWrite(LED_BUILTIN, LOW);

  delay(5000);                          // Wait a few seconds
}
```

**What I get** when running the sketch:

```
ecimal: 6247424 (24Bit) Binary: 010111110101010000000000 Tri-State: FF11FFF00000 PulseLength: 352 microseconds Protocol: 1
Raw data: 10940,348,1088,1044,388,340,1088,1044,384,1048,384,1044,380,1044,388,1044,388,336,1088,1040,384,344,1088,1040,388,328,1096,1040,380,340,1088,336,1088,336,1088,336,1084,336,1084,332,1088,332,1092,336,1084,332,1084,340,1084,
```

It doesn't matter if I try to make the PulseWidth shorter (144) or longer, it's always 352 ms.

Do you have an idea what's going wrong?

fingolfin, 2017-02-10:
You have to set the protocol first, and *then* the pulse length. The reason is that setting the protocol also sets the pule length to the default for that protocol, so you need to change the pule length afterwards.

You can also do `mySwitch.setProtocol(1, 144);` to set both at once.

quaec, 2017-02-10:
Thanks a bunch, now it works!

Would it be a good idea to change the order in the demo file? For the case that somebody uncomments both the protocol line and the PulseLength line.

closed 2017-02-10 by fingolfin