sui77/rc-switch #125: NodeMCU ESP8266 repeat my send signal twice causing light to tun "ON" then "OFF" instantly.

pico563, 2017-02-23:
Hello all

```
 mySwitch.enableTransmit(13);  //nodeMCU GPIO13 --->D7
// set protocol (default is 1, will work for most outlets)
/* mySwitch.setProtocol(1)  */
  mySwitch.setProtocol(1,304);

 // Optional set pulse length.
 //mySwitch.setPulseLength(320);

 // Optional set number of transmission repetitions.
 mySwitch.setRepeatTransmit(4);
```
```
BLYNK_WRITE(V52)
{
int pinValue = param.asInt(); // assigning incoming value from pin Vc to a variable
if (pinValue==1)
mySwitch.send("010011110001000000110000"); // DINING ROOM   B  (Medium)
Serial.println("V52:DINING ROOM   B  (Medium) ON/OFF ");
}

```
here is the serial print for any button i pressed.

V52:DINING ROOM   B  (Medium) ON/OFF 
V52:DINING ROOM   B  (Medium) ON/OFF 

I tried changing transmission repetitions ,  pulse length  but no hope.
always i got the signal transmitted twice.

thanks

`

fingolfin, 2017-02-27:
So you have a switch which uses the *same* signal for turning someting on *and* off? That's somwhat unusual, as most systems rely on repeating the transmission multiple times to deal with signal losses.

Anyway, you hav ` mySwitch.setRepeatTransmit(4);` in you sample code. Did you try it with 1 instead of 4 (you said you "tried changing transmission repetitions", but it is not clear from that what exactly you tried).

pico563, 2017-02-28:
When I tried 1 the 315Mhz  receiver connected to my nodeMCU (ESP8266)  serial print reported that two repeated signal was sent but unfortunately  no response from the hardware.
same with 2.
with 3 it works but few times i got no response from the hardware.
the best choice was with 4  but still light goes on then off (with that type of HW  switch ) 
in all 4 serial print reported that two repeated signal was sent.

```
Decimal: 3952832 (24Bit) Binary: 001111000101000011000000 Tri-State: 0110FF001000 PulseLength: 302 microseconds Protocol: 1
Raw data: 9372,318,894,320,892,928,287,927,288,923,289,926,289,317,898,314,899,314,898,922,293,314,898,922,293,314,894,316,896,318,895,317,899,921,294,921,294,200,1011,314,897,316,894,318,899,313,894,316,901,

Decimal: 3952832 (24Bit) Binary: 001111000101000011000000 Tri-State: 0110FF001000 PulseLength: 302 microseconds Protocol: 1
Raw data: 9372,318,894,320,892,928,287,927,288,923,289,926,289,317,898,314,899,314,898,922,293,314,898,922,293,314,894,316,896,318,895,317,899,921,294,921,294,200,1011,314,897,316,894,318,899,313,894,316,901,

```
here i must mention that when i use the original remote i got one signal sent  with serial print.

thanks