sui77/rc-switch #127: Reduction in Range

fishersolutions, 2017-02-27:
I have found that the range of detection code is lower when using RC Switch on a esp8266 compared to the ebay relay modules.

I have even taken the RF module from a working Relay module and tried it on the ESP9266 and RCSwitch.
The results is the range drops from 20M+ to no more than 3m.

The TX is the same.  The ebay KEY FOBS.  24bit Proto 1.

Anyone else seeing reduce range?
Any ideas?
I have tried replacing the default Antenna with only slight improvements and nothing like the relay module range

fingolfin, 2017-02-27:
The range depends entierly on your transmitter module and how much output power it has, as well as the size (and shape etc., as usual) of the antenna. Nothing rc-switch does should affect it.

closed 2017-02-27 by fingolfin

fishersolutions, 2017-02-27:
Hi Fingolfin,
The problem I have with that statement is my hardware is identical.
I have take the RX module off 1 of 4 relay on/off modules and placed it on the ESP8266 and rcswitch.
I then place another relay module in the same location as ESP8266 setup.
Using the same RX FOB the Original Relay module has much better range than the ESP8266.
So this is a back to back test.

So the issue I am facing is related to the ESP8266,  I have bypass Cap on the RXmodule just in case.
A quick look at the original RX circuit indicates it just feeding into a uP.

Is there any filtering in the rc-switch?
Given rcswitch is universal is there some loss vs an algorithm that know what to expect.
 
Thanks in Advance

fingolfin, 2017-02-27:
rc-switch simply toggles a digital IO pin on and off, toggling it between 0 volts and whatever the output voltage of your controller is (i.e. typically 3.3V or 5V). No "filtering" or anything otherwise fancy is don. You can also verify this by lookubg at `RCSwitch::transmit`.