sui77/rc-switch #128: Receiving causes to much load for Arduino

bartrouw, 2017-03-09:
Thanks so far for all the good work!

I am building a (incandescence) light dimmer which i want to control remotely. The dimmer is using two interrupts: one hardware interrupt for the zerocross detection, one hardware timer for the triac pulse which lighten the lamp a couple of microseconds after the zerocross detection.
[lightdimming using arduino](https://arduinodiy.wordpress.com/2012/10/19/dimmer-arduino/)

This time specific sequence is disturbed when I call enableReceive, even when I don't send anything, which causes the light bulb to start flickering. 

Are there ways to make the code less resource intensive? 
I was thinking about the following:
1.  Make it possible to receive only for a specific protocol.
2. Take receiveProtocol out of the handleInterrupt function to make the handleInterrupt function more lightly.

Is this possible?

bartrouw, 2017-03-10:
I did some coding myself.

1. I added a setReceiveProtocol function. So receiving can be done for only a single protocol.
2. I moved receiveProtocol out of handleInterrupt and put it in available. So it the received value is decoded on demand.

This speeded things up! Light bulb stopped flickering. I don't know if the changes I made are good for everyone but I'm very happy with it.
[rc-switch.zip](https://github.com/sui77/rc-switch/files/835049/rc-switch.zip)

AhMaizeBalls, 2017-08-06:
You absolute badass! I've been struggling with inconsistent received values but now with your addition and my own protocol it's rock solid! Thanks!