sui77/rc-switch #129: Mega 2560 support

evilgreen, 2017-03-15:
Hello
The examples provided seems to be not operational on Arduino Mega 2560 - there is no output regardless of the pin/interrupt used. The hardware has been checked of course.

tunjid, 2017-03-23:
@evilgreen I've used this project on an Arduino 2560 with no problems. What pins are you using?

evilgreen, 2017-03-23:
@tunjid I've tried various pins. You can see the whole story [here](https://github.com/1technophile/OpenMQTTGateway/issues/22).
Thank you!

tunjid, 2017-03-24:
Are you using a library in your project that uses one of the Arduino's timers internally? If you are and the interrupt pin you're using uses the same timer you'll likely run into issues. An example of a libray that depends on the timers would MsTimer2 or FlexiTimer2.

Try using pin 3 on the Mega for interrupt receiver pin.

```C
// The interrupt pin for the RcSwitch. See http://arduino.cc/en/Reference/AttachInterrupt
#define INTERRUPT_RX digitalPinToInterrupt(3)
```

evilgreen, 2017-03-24:
As I mentioned in the thread I referenced earlier, even the examples provided with the library are not operational. Tested with Uno - same story. At the same time all the boards and a radio are working, various interrupt checking sketches are working and even [SimpleRcScanner](http://test.sui.li/oszi/Sketchbook/SimpleRcScanner.ino) works. Nothing works when I'm using the rcswitch library.

tunjid, 2017-03-24:
This is the [sketch](https://codebender.cc/sketch:80286#RCSwitch%20-%20Sniffer.ino) I use with my Mega and it works fine.

Try it again, but change the pin you are using for an interrupt pin and see it it helps you. Replace 

```C
// The interrupt number, on the MicroView this is PIN 11. On Arduino it is PIN 2. See http://arduino.cc/en/Reference/AttachInterrupt
#define INTERRUPT_RX 0
```

with this and use pin 3. If that doesn't work, try it again with the valid Mega pins [here](https://www.arduino.cc/en/Reference/attachInterrupt), switching the value passed to digitalPinToInterrupt with the pin you're using.

```C
// The interrupt number, on the MicroView this is PIN 11. On Arduino it is PIN 2. See http://arduino.cc/en/Reference/AttachInterrupt
#define INTERRUPT_RX digitalPinToInterrupt(3)
```

evilgreen, 2017-03-24:
Does not work as before. I'll try to install IDE on another PC during weekend and try again. Thank you, @tunjid

1technophile, 2020-01-05:
I'm closing this issue as I can confirm that RCswitch work on arduino MEGA

closed 2020-01-05 by 1technophile