sui77/rc-switch #133: ESP32 sometimes it receives data but after some time it stops

MacWyznawca, 2017-04-09:
ESP32 (ESP-WROOM-32 dev board) sometimes it receives data but after some time it stops while transmission operates in the same application all the time.
I checked connections on different pins ...
The same code running on the ESP8266 (Electrodragon relay) works fine.

Any hint where to look?

JOL2912, 2017-04-10:
Please post the code example.

MacWyznawca, 2017-04-11:
@JOL2912 

All code: [http://macwyznawca.pl/main.ino.zip](http://macwyznawca.pl/main.ino.zip)

Fragments with reverences to RF:

`#include <WiFi.h>
#include <PubSubClient.h>
#include <RCSwitch.h>

[…]

void setup() {
  Serial.begin(115200);
  setup_wifi();
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);

  for (byte i=0; i < BUTTONS_NUMBER; i++ ) {
    if (buttonPinOuts[i]<99){
      pinMode(buttonPinOuts[i], INPUT_PULLUP);
    }
  }

 myRfSwitch.enableReceive(RFRECIEVEPIN);
 myRfSwitch.enableTransmit(RFTRANSMITPIN);
}

void loop() {

  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  if (myRfSwitch.available()) {
    unsigned long rfButton = myRfSwitch.getReceivedValue();

    Serial.print("Guzik RF: ");
    Serial.println(rfButton);

    for (byte i=0; i < RF_BUTTONS_NUMBER; i++ ) {
      if (rfCodes[i] > 0){
        unsigned long currentMillis = millis();
        if ((currentMillis - previousMillis >= interval) && (rfCodes[i] == rfButton) && (rfButton != lastRFbutton)) {
          if (client.connected()) {
            for(byte j = 0; j<RF_BUTTONS_COMMANDS_NUMBER;j++){
              if(rfButtonsTopic[i][j] != ""){
                client.publish(rfButtonsTopic[i][j], rfButtonsTopicCommand[i][j]);
                Serial.print("Topic: ");
                Serial.println(rfButtonsTopic[i][j]);
                Serial.println(rfButtonsTopicCommand[i][j]);
              }
            }
          }
          lastRFbutton = rfButton;
          previousMillis = currentMillis;
        }
      }
    }
    myRfSwitch.resetAvailable();
  }
}`

JOL2912, 2017-04-24:
Code seems OK.  I have the same problems using my ESP32 Dev Module.

p3p1, 2017-06-18:
I have the same problem with similar code. Do you have fixes?

fpupor, 2017-08-30:
Same problem... =(

1technophile, 2020-01-05:
With the below ESP32 arduino environment:
espressif32@1.11.1
Arduino ESP32 1.0.4
I get stable ESP32 RF receiving on OpenMQTTGateway.
So I close the issue.

closed 2020-01-05 by 1technophile