sui77/rc-switch #135: Receiving data from unknown devices

fntslz, 2017-04-15:
Hello,
I 2 remote controls 433, one is detected as Protocol No. 1 while the other is not detected (has a coding with 12dip-switch).
How can I do to be able to receive the 12 dip-switch?
Thank you.

Martin-Laclaustra, 2017-04-15:
check issues:
https://github.com/sui77/rc-switch/issues/103
https://github.com/sui77/rc-switch/issues/112

Your second control may be 32 bits constant bit-length but with different pulse lengths (like all the standards in rc-switch) or a new NEXA-like 64 bits different-on-off-bit-length. You will be able to see this when you record your signal.
It can help if you declare which platform you do use (Arduino or Rpi).

If you use Arduino, you can use Stuat's signal recorder:
https://github.com/sui77/SimpleRcScanner

fntslz, 2017-04-15:
yes..i use arduino with cheap rx433 .
alfter i try to recorder.
thanks.

fntslz, 2017-04-15:
OK done.
Now you can explain to me how to create my protocol?
Thank you.
[image]

Martin-Laclaustra, 2017-04-16:
It looks like a 12-bits inverted protocol.
(I am not very familiar with inverted protocols)
Your code is either:
010100110100
or
101011001011
(one of these should match your switch, please report which)
Your protocol:
{ 330, { 34, 1 }, { 1, 2 }, { 2, 1 }, true }

Use:
setProtocol(the order number where you insert it in the RCSwitch.cpp file)
and
send(1332, 12)
or
send(2763, 12)

Please report if you succeed and describe in detail your brand and model. Maybe even with photographs or links to seller.
It would be great if your could open your original remote and report which chipset it uses.

fntslz, 2017-04-16:
very great ...

it works.
the model is a "FAAC" fixed code, in Italy it is very common.

Please explain to me how did you determine the protocol?
{330, {34, 1}, {1, 2}, {2, 1}, true} what do they mean?
I could not understand ...

Thank you for your support ...
[image]

Martin-Laclaustra, 2017-04-16:
Check:
overall explanation of the signal:
https://github.com/sui77/rc-switch/wiki/KnowHow_LineCoding#ev1527-rt1527-fp1527-or-hs1527
how the signal is encodes:

fntslz, 2017-04-16:
Hello,
I'm sorry ... I read the first link ..but I have not understood ... maybe I'm stupid ...
Kindly you explain it to you with the colors?
[image]

how you do the calculations?
I can not understand, seeing the graph, to build the protocol ...

Can you explain to me step by step how you do?

Thank you very much,

Martin-Laclaustra, 2017-04-16:
[image]
@sui77 you might consider using this image in the wiki.
Best

fntslz, 2017-04-16:
..I'm sorry ...
but still I can understand with the "330", how do you calculate ... ??
I made (400 + 612 + 744 + 264 + 400 + 612) / 6 = 505 ... definitely wrong ..where?
Excuse me if I insist ... but I want to understand .. thanks.

Martin-Laclaustra, 2017-04-16:
400 + 612 = 1012
744 + 264 = 1008
400 + 612 = 1012
752 + 260 = 1012
Every bit lasts aprox. 1012 us. Every bit is made of 2 units, one long and one short. It is unlikely that the long-short ratio is different (as data suggests) between on and off bits. Most probably it is the same. Let's estimate that ratio combining long lengths and short lengths of a "0" and a "1" type bits: (612 + 752) / (400 + 260) = 2.07
So each bit has 3 timing "ticks", 2 for the long part and 1 for the short part.
1012 / 3 = 337 ~ 330
An approach considering ratios of 1.5 for "0"s and 3 for "1"s would have made quite complex to find a minimum common pulselength for both. It would not have made sense that engineers  would have designed the remote that way.

fntslz, 2017-04-17:
OK thanks for the help.
Now I have a transmitter, no brand, which transmits approximately every three seconds a string type here ...
honestly can not costriure the protocol ... can you do it?
... unfortunately I have difficulty to understand how it works ...
maybe it's my limit ...
Thanks again for the availability
[image]

kiralikbeyin, 2017-04-24:
@fntslz #103

fntslz, 2017-04-26:
Hello guys,
I've been trying to get my tx signal for days.
Unfortunately my tx is a rolling code ...
The first 2 bytes are always the same .. the remaining length varies ...
How could I get it to receive?
Ideas or suggestions?

Thank you.

rafkacz, 2017-04-26:
Hi - can you help with decoding signal for MCE07 rc switch?
This is 4 channel switch. Below signals for ch1 off and ch1 on.
First 10000us is added by me manually. 
[image]
[image]

fntslz, 2017-04-27:
Hello ...
I can not help you.
We look forward to help from Martin-Laclaustra .
Bye

rafkacz, 2017-04-28:
I managed to decode it myself (MCE07). 
proto = { 350, {  10,  10 }, {  1,  1 }, {  1,  3 }, false }

It occured that fst 4 pulses can be ignored (when sending data). Code length is 41 bytes - last 9 bytes codes channel no and function. 

There is problem with sending data due to 32 bit limitation of long used by RcSwitch. 
I added one func that sends code from char* without conversion to long. 

```
	void RCSwitch::sendB(const char* sCodeWord) {
		
		if (this->nTransmitterPin == -1)
		return;
		
		#if not defined( RCSwitchDisableReceiving )
			// make sure the receiver is disabled while we transmit
			int nReceiverInterrupt_backup = nReceiverInterrupt;
			if (nReceiverInterrupt_backup != -1) {
				this->disableReceive();
			}
		#endif
		this->transmit(protocol.syncFactor);
		for (const char* p = sCodeWord; *p; p++) {
			
			if (*p != '0')
			{
				this->transmit(protocol.one);
			}
			else
			{
				this->transmit(protocol.zero);
			}
		}
	}

```

Its almost like RCSwitch::send(const char* sCodeWord) but without conversion to long and sync signal moved at begining of message. 

The codes are:

```
                 1         2         3         4
       01234567890123456789012345678901234567890
                                       XXXXXXXX
Ch1 on 00101001011000001101011010011111100001111
Ch2 on 00101001011000001101011010011111010010111
Ch3 on 00101001011000001101011010011111110000111
Ch4 on 00101001011000001101011010011111001011011
Ch* on 00101001011000001101011010011111101001011

Ch1 on 00101001011000001101011010011111011110001
Ch2 on 00101001011000001101011010011111101101001
Ch3 on 00101001011000001101011010011111001111001
Ch3 on 00101001011000001101011010011111110100101
Ch* on 00101001011000001101011010011111010110101
```
Bits marked by X at top - codes channel no and function (fst 4 bits read from right to left - codes channel no - next 4 bits are negation of previous 4 bits). 
OFF codes are negation of On code. 
The last one bit - is the question - i had to add it to make it work... 

Sending data works but - it is impossible to receive codes generated by original MCE07 RC due to fst 4 starting sync pulses.

Any ideas how to make receiving possible (how to filter out this 4 pulses starting sync signals)?

Martin-Laclaustra, 2017-04-30:
I am glad that you are able to send signals.
Sorry it is not easy to adapt rc-switch for protocols with something different from a single-sync-gap.

eibt, 2017-07-28:
Hi, so I got this remote from a car security system:

[image]

I wonder if it's possible to read this remote from an Arduino using **rc-switch**, I used the **ReveiceDemo_Advanced** sketch but it didn't recognize the protocol.

I opened the remote and it has this IC that reads **PT2240B-S**:

[image]

I used the **SimpleRcScanner** and I got this when pressing the lock button:
🔽🔽🔽
[image]
🔼🔼🔼

After looking at the image for some time I thought it could be interpreted like this? that would make a 24 bit code word:

[image]

Could you help me define a protocol that would correctly interpret this remote? I tried the following but it didn't work:

`{ 900, {  2, 23 }, {  2,  2 }, {  6,  1 }, false }`

I tried applying subtle changes to those values without success, maybe I'm not calculating the pulse length correctly?

DFW-Noob, 2018-06-21:
Hello i'm not sure if i am doing this right or how to get all the info i need from the signal like Pulse and snyc info from this but i think i got the Binary right can some one look and give me the bit of info i might be missing 
[image]

Martin-Laclaustra, 2018-06-21:
Read:
https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_1

Anyways, it seems that you will not be able to use rcswitch to receive this remote, because it uses 2 sync bits. Also, for sending commands you will need to send a pre-programmed sequence of timings. You will not be able to use the single number that you decoded.

My branch [transmittimingsarray](https://github.com/Martin-Laclaustra/rc-switch/tree/transmittimingsarray) described in https://github.com/sui77/rc-switch/pull/170 , https://github.com/sui77/rc-switch/issues/146 , and https://github.com/sui77/rc-switch/issues/163#issuecomment-330576310
is capable of sending an array of pulses and will probably be able to control your sockets or receivers.

Martin-Laclaustra, 2018-06-22:
You can use the protocollessreceiver branch (you get details on how to download it correctly here: https://github.com/sui77/rc-switch/issues/103 ) but it will be of no use for your remote.
Follow the links in my previous reply if you want to reach practical advances.

DFW-Noob, 2018-06-22:
Thanks i'm really trying to follow along and but keep hitting a wall. with the output i first posted i noticed there was the same code for the level i was looking for Example 
one decode i did i got 
1100000010000000101001110101100011101111100
and it had a level setting of 99 
110000001000000010100111010"1100011"101111100

what i am trying to do is capture the signal from a dog collar using an arduino like this 
https://gist.github.com/tkuester/67f2d8f5c03aee22c6d7

 |    ?   |    ?   |    ?   |  level |   ?    |
 |        |        |        |00010100|        |
0|10001000|01101010|10000000|00010100|11101110|0 <-- Shock?
0|10000001|01101010|10000000|00010100|01111110|0 <-- Vibrate?
0|10000100|01101010|10000000|00010100|11011110|0 <-- Blink?
0|10000010|01101010|10000000|01100100|10111110|0 <-- Beep?

so my output looks right i'm just not sure what and how to fully decode them and rebrod cast them back

Martin-Laclaustra, 2018-06-23:
Sorry. This is completely unrelated to this issue.
You can not use rcswitch to decode or rebroadcast codes for your signals. It is incompatible.

stich86, 2018-08-13:
Hi Martin,

can you help me to decode my remote? It's a FAAC @315mhz, but should be fixed code on the receiver side (because each remote it's programmed from it)

here is the three sample that I got (i've tried various time - pressing and hold the button at number 2 on Arduino serial - but don't have flat line at the start)

[image]

Hope you can help me.

Thanks in advance

stich86, 2018-08-15:
Hi Martin,

Thanks for your response. So it’s not a 2-bit sync protocol? Because I’ve seen two times a big number before the sequence of command.

I’ll try today to add the suggested protocol on the cpp file and test it.

I let you know

Thanks again!

stich86, 2018-08-15:
It doesn’t seem to work.
I’ve added the suggested code to RCSwitch.cpp and loaded the ReceiverDemo on the Arduino, but pressing button on my remote doesn’t show anything on the serial monitor :(

stich86, 2018-08-15:
i'm still not understand how do you have calulated that data. The wiki is not so clear.. also the example and pictures reports different numbers.

Martin-Laclaustra, 2018-08-16:
1. Read the wiki links above. Do ALL that it says there before posting again.
2. Learn how long is your code.
3. Translate your signal in 0s and 1s, manually. (I already gave you the decoding interpretation in my previous post).
4. Try to replicate transmission before trying to receive.
5. If any, post numbers, not images.

stich86, 2018-08-17:
Hi Martin,

I've readed the wiki but some parts are not very clear like 15th.

I've sample five times my code, this is one of the sample that looks like the other one:

`820,808,844,780,840,1604,1648,1596,1656,1592,1632,1620,1656,3216,3280,1596,1636,1616,1656,784,828,796,844,1592,1652,1600,1636,1612,1660,1600,1652,776,836,800,832,1600,1628,1616,1660,780,828,800,840,1596,1656,780,848,780,824,1612,1660,1600,1648,780,844,792,832,1596,1640,1612,1652,1608,1644,780,844,792,828,788,844,780,828,1612,1656,1600,1652,1592,1648,1600,1644,3232,3280,1600,1644,1604,1636,812,832,788,840,1600,1644,1596,1648,1604,1640,1612,1652,784,824,800,848,1592,1644,1608,1640,808,832,784,844,1600,1652,776,848,788,824,1604,1648,1604,1656,780,820,808,840,1596,1648,1604,1644,1608,1656,780,820,804,844,780,848,784,828,1604,1644,1608,1660,1604,1640,1592,1648,3236,3280,1604,1648,1592,1640,800,844,780,836,1600,1656,1608,1644,1596,1644,1608,1648,800,828,788,844,1604,1648,1592,1640,800,840,784,840,1596,1660,780,820,812,840,1592,1640,1612,1652,796,832,784,844,1600,1652,1596,1636,1616,1660,792,824,792,840,784,828,804,840,1592,1640,1612,1660,1592,1656,1596,1656,3232,3276,1596,1652,1596,1652,784,844,784,824,1612,1656,1604,1648,1596,1664,1592,1632,808,840,788,840,1604,1648,1596,1656,780,848,780,820,1620,1656,780,832,796,844,1592,1656,1596,1632,808,840,784,848,1604,1644,1596,1656,1596,1636,808,836,784,844,784,824,800,844,1592,1660,1592,1636,1616,1660,1604,1648,3220,3260,1616,1656,1608,1648,780,840,800,820,1600,1636,1616,1660,1604,1648,1592,1648,792,840,784,828,1612,1656,1604,1644,784,840,796,820,1604,1640,808,832,784,844,1600,1652,1596,1644,788,848,780,828,1612,1656,1600,1652,1592,1644,792,848,776,828,808,836,784,844,1604,1644,1592,1648,1608,1644,1608,1652,3224,3268,1608,1652,1600,1656,780,828,804,840,1592,1644,1608,1652,1600,1656,1604,1648,780,848,784,824,1612,1640,1608,1660,776,824,812,836,1592,1640,800,840,784,836,1604,1652,1604,1648,784,840,784,824,1612,1648,1604,1656,1600,1652,780,840,784,828,800,840,784,844,1596,1656,1600,1652,1592,1640,1612,1656,3220,3280,1596,1636,1616,1656,796,824,788,852,1592,1652,1596,1636,1616,1668,1588,1656,776,832,804,840,1588,1644,1608,1664,784,828,792,844,1592,1660,780,844,780,828,1612,1660,1600,1652,776,840,800,832,1596,1636,1612,1660,1600,1656,776,836,800,832,784,848,776,832,1608,1660,1604,1648,1592,1660,1596,1632,3240,3288,1592,1656,1596,1636,808,840,784,848,1600,1648,1592,1656,1592,1636,1620,1660,776,824,800,844,1592,1656`

transpose this code on XLS (and in the site that create the wave), show me there are only two high number (> 3000) and two groups formed by numbers that are between 1500-1700 and 700-900. The first thing that i'm missing is the pulse lenght (it's not specified very well on the wiki how to obtain). Look at your suggestions, the number is the average from the big number after the sync bit and the small number. It give me about 804, that you have translated to 800.

Regarding sync bit, in your suggestion you have put "4, 4", that look at the wave seems right (because a single wave seems formed by 4 small waves). For 0s bit you have put "2, 2" that like the sync bit are formed by 2 small waves on up and on down (the first two "1600" values), the last is 1s bit that seems formed just only a single small wave (and you have put "1,1").

So what i'm missing? Until the end of next week I cannot test this code because i don't have another Arduino here with me or my gate to open. 
But I was surprised that rc-switch library cannot detect the binary code 

Thanks in advance for any help.

Martin-Laclaustra, 2018-08-17:
I updated the wiki:
**Pulse length** (in this library) is a concept that allows simplifying the "formula" for the encoding: It is roughly the [greatest common divisor](https://en.wikipedia.org/wiki/Greatest_common_divisor) of the different durations identified previously. That way, those durations can be described with simple integer small numbers, multiples of the pulse length. In practice: in excel: a) sort the durations; b) identify the groups; c) use the function "average" to average the durations within each group; d) approximate the greatest common divisor; e) check that the pulse length multiplied by the simple numbers roughly reproduce the initial timings.

Apply that to your numbers. pulse length: 813 (roughly the one I approximated: 810). (averages of the groups: 812.9, 1625.5, 3252)

What you have to do next?
- Your code is 32 bits length (2. in my previous post)
- Decode manually (3. in my previous post)
- Mimic the signal (4. in my previous post)... and tune it until it works.

stich86, 2018-08-25:
ok, 

finally i got working these damn modules. Checking the XLS, my code seems to be 818780400. I've send it (after add protocol on *.cpp file) and generate a new wave on the test site, but it seems missing the sync bite. 

Here is the code generated by SimpleRCScanner

`1660,792,856,784,856,788,856,784,860,1588,1656,1592,1660,1592,1656,1592,1696,1600,1672,1588,1676,784,868,784,876,1588,1684,1584,1676,1588,1672,1588,1676,776,876,776,868,1592,1672,1584,1676,780,872,772,876,1580,1676,780,864,780,868,1584,1668,1588,1668,780,872,776,872,1580,1680,1576,1672,1580,1672,780,864,776,868,776,864,776,860,1584,1680,1572,1672,1580,1672,1580,1712,1584,1684,1580,1684,772,888,768,880,1580,1684,1576,1684,1572,1684,1580,1680,776,888,768,876,1576,1684,1576,1676,780,876,772,876,1580,1684,768,876,768,880,1572,1680,1576,1684,768,872,772,876,1576,1672,1580,1680,1576,1680,772,872,768,872,776,872,772,876,1568,1680,1576,1676,1572,1680,1572,1720,1576,1688,1576,1688,768,884,768,884,1576,1688,1576,1688,1572,1684,1576,1684,772,884,772,880,1572,1688,1572,1684,772,876,768,876,1576,1680,768,880,772,880,1572,1688,1568,1680,772,880,764,876,1572,1688,1568,1680,1572,1684,764,880,768,872,768,880,760,876,1572,1680,1568,1680,1572,1676,1576,1724,1572,1688,1572,1688,772,884,768,888,1572,1696,1572,1692,1576,1688,1572,1700,764,880,772,892,1560,1692,1568,1688,768,884,764,880,1568,1688,764,888,760,888,1568,1688,1568,1684,764,884,764,876,1572,1684,1568,1684,1568,1684,760,884,760,884,760,880,760,880,1568,1680,1568,1676,1576,1680,1572,1716,1576,1688,1572,1696,768,884,768,884,1568,1696,1564,1692,1568,1696,1568,1688,764,884,764,888,1568,1688,1576,1688,768,884,764,884,1572,1688,768,880,760,888,1564,1688,1572,1688,760,884,768,880,1572,1688,1568,1684,1576,1680,768,880,768,876,764,876,764,876,1572,1676,1576,1676,1576,1684,1568,1720,1580,1692,1568,1696,764,884,768,884,1572,1688,1572,1688,1572,1696,1564,1692,764,888,760,892,1564,1688,1572,1692,764,884,764,888,1568,1688,768,884,760,888,1564,1688,1568,1688,760,880,768,884,1568,1688,1572,1684,1568,1684,764,884,764,884,756,880,764,884,1560,1680,1576,1692,1556,1680,1572,1728,1568,1692,1572,1692,768,888,764,884,1576,1692,1568,1696,1572,1692,1568,1688,768,892,760,896,1560,1692,1568,1696,760,888,764,884,1568,1692,756,892,760,880,1572,1688,1568,1688,764,880,768,884,1564,1684,1568,1688,1572,1680,764,880,768,880,760,876,764,880,1568,1684,1568,1680,1572,1684,1564,1728,1564,1696,1568,1704,756,884,772,884,1568,1696,1568,1696,1568,1692,1568,1692,764,892,760,892,1564,1688,1572,1692,764,884,764,884,1572,1688,764,884,768,884`

and this is the sketch used:

`/*
  Example for different sending methods
  
  https://github.com/sui77/rc-switch/
  
*/

#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {

  Serial.begin(9600);
  
  // Transmitter is connected to Arduino Pin #10  
  mySwitch.enableTransmit(10);
  
  // Optional set protocol (default is 1, will work for most outlets)
  mySwitch.setProtocol(11);

  // Optional set pulse length.
  // mySwitch.setPulseLength(810);
  
  // Optional set number of transmission repetitions.
  // mySwitch.setRepeatTransmit(15);
  
}

void loop() {

  /* Same switch as above, but using decimal code */
  mySwitch.send(818780400, 32);
  delay(300);  
  mySwitch.send(818780400, 32);
  delay(300);  
  mySwitch.send(818780400, 32);

}`
what i'm missing?

Thanks!

Martin-Laclaustra, 2018-08-26:
Replace:

`
mySwitch.setProtocol(11);
`
With:

```
static const RCSwitch::Protocol customprotocol = { 810, { 4, 4 }, { 2, 2 }, { 1, 1 }, false };
mySwitch.setProtocol(customprotocol);

```

... and report again your recorded timings.

stich86, 2018-08-26:
Ok using your suggestion, I wa able to mimic the same signal that also opened the gate.

Now I need to understand why it doesn’t work adding to the *.cpp file.

Is it possible to use this library on a raspberry Pi and send the same command?

Thanks

Martin-Laclaustra, 2018-08-27:
> Is it possible to use this library on a raspberry Pi and send the same command?

Yes, use 433Utils.

typh289, 2020-05-17:
> It looks like a 12-bits inverted protocol.
> (I am not very familiar with inverted protocols)
> Your code is either:
> 010100110100
> or
> 101011001011
> (one of these should match your switch, please report [...]

It's from some years ago but in case it come useful to anyone (as this did to me :) )
I can confirm this protocol work with common (in Italy)  FAAC remotes (ES. tm433)
The right code in the quoted case is the first one eg. 010100110100

Another protocol that worked out on a different (but similar) gate remote is:
{ 780, { 6 ,  1 }, {  1,  2 }, {  2,  1 }, true }
Will update also this with make and model, as i derived it from a generic replacement remote, but by adding this protocol in the library and testing with the advanced or simple receive demo may be already useful to someone with similar remotes i guess

mellbo, 2020-05-24:
I need help with my unknow protocol. Please help.
SimpleRCScanner:

**Button1**
`868,424,864,432,856,436,428,860,876,416,880,416,868,424,432,852,872,420,876,420,868,428,436,848,444,848,464,824,452,840,472,808,25772,2612,852,440,852,440,848,440,852,440,852,440,420,864,864,432,860,428,852,440,852,436,852,440,852,440,420,864,856,436,432,856,444,844,872,420,444,844,444,840,888,412,452,836,456,832,464,828,916,376,884,412,884,408,448,840,456,828,888,412,872,420,444,836,888,408,888,408,876,412,872,420,864,428,860,432,864,428,856,436,432,852,872,416,880,412,868,428,428,860,868,420,868,428,856,436,436,848,444,844,456,832,460,832,460,828,25772,2608,852,444,852,436,856,440,852,440,848,444,420,868,856,436,856,436,852,440,856,436,852,444,848,444,420,864,860,432,436,856,440,848,872,424,436,848,448,844,876,416,468,824,468,820,464,828,896,392,884,416,876,412,456,840,448,836,880,416,884,412,444,844,880,416,868,424,872,420,868,424,868,428,872,420,860,428,860,436,428,856,868,424,880,412,868,424,444,848,880,412,864,432,868,420,436,848,448,844,456,832,456,832,460,828,25820,2616,848,448,844,448,844,444,848,444,848,444,420,864,864,428,864,432,856,432,856,436,852,440,848,444,424,864,860,428,428,856,440,852,872,420,436,848,452,840,876,416,452,840,460,824,476,820,888,400,892,404,892,404,448,840,460,824,888,408,900,396,456,840,876,416,880,412,872,424,868,428,868,428,864,424,872,428,856,436,432,852,872,420,868,428,864,428,440,852,872,416,872,424,872,424,428,860,448,840,452,840,468,820,460,832,25804,2608,852,440,852,444,852,440,852,440,856,440,420,868,860,432,856,436,856,440,856,436,856,440,852,436,428,864,864,428,432,860,440,848,872,428,440,848,448,840,884,416,456,836,476,820,460,828,896,400,892,404,880,416,468,824,460,828,888,412,884,412,444,848,888,404,880,416,884,412,872,424,868,424,864,432,860,436,872,416,436,856,876,416,868,432,864,428,432,852,884,416,872,420,872,424,432,856,448,844,448,844,460,832,464,828,25824,2612,856,436,856,444,852,440,852,440,848,448,420,868,860,436,860,432,856,444,852,440,852,440,852,440,428,860,864,432,428,860,440,848,868,432,436,852,448,844,876,420,456,832,
`

Button2
`416,868,432,864,860,436,428,856,440,856,872,424,448,844,444,844,456,840,880,420,868,424,868,424,436,856,444,844,880,416,868,424,872,424,440,852,868,424,868,432,860,432,856,440,860,436,848,440,856,444,420,868,860,432,432,856,440,848,872,428,432,856,444,852,868,424,868,428,440,848,872,432,432,856,880,416,25800,2628,844,452,844,444,840,452,840,456,836,452,412,876,848,444,848,448,844,448,844,448,840,452,844,444,420,868,856,440,424,864,432,860,864,428,432,856,444,852,864,428,440,848,452,836,460,836,884,412,876,412,872,428,436,856,440,848,872,424,872,424,860,436,428,852,876,420,864,432,860,436,852,436,868,424,856,444,848,440,428,860,860,432,440,848,436,852,872,424,440,848,448,844,876,420,872,424,432,856,872,420,440,856,868,424,25764,2624,844,448,840,448,844,452,840,448,844,448,412,876,848,444,848,440,852,444,844,448,844,440,848,448,412,876,848,444,424,860,432,860,864,428,432,856,440,852,872,424,436,852,452,836,456,836,880,416,872,420,876,420,436,852,448,844,880,420,864,432,860,436,428,856,872,428,860,436,860,428,860,440,852,440,848,448,856,440,420,864,860,432,428,864,432,852,872,424,440,852,444,848,876,408,876,424,436,852,872,424,440,852,872,420,25764,2624,844,448,840,452,844,448,848,448,836,456,412,876,848,440,852,440,844,444,848,448,844,448,840,448,416,868,856,436,424,860,432,860,864,428,436,852,444,848,876,424,440,848,448,844,448,844,884,408,876,420,868,428,436,856,440,848,876,420,860,436,860,432,428,856,864,428,860,432,860,436,852,440,856,436,852,440,848,444,416,868,856,436,428,860,440,848,868,428,436,852,444,848,876,420,864,424,444,852,872,420,440,852,872,424,25776,2624,840,452,844,452,840,452,844,448,844,448,416,876,848,444,848,448,844,448,844,444,848,444,844,452,412,872,852,440,424,860,432,860,864,428,432,856,436,848,876,424,440,848,444,844,452,836,888,412,876,416,864,428,440,848,444,844,872,420,868,424,868,428,432,856,864,428,860,428,864,436,852,440,852,436,856,436,848,444,420,868,856,436,432,856,440,852,868,420,440,852,444,844,872,420,872,424,436,852,872,424,440,844,872,424,
`

LeandroAR, 2020-10-18:
Hi Martin,

I have a rollshutter with a AC123-01 433MHz remote, up+stop+down buttons. I understand this is not a single-sync-gap, therefore I can't decode it using rc-switch in Arduino. I was able to get some values from SimpleRcScanner, should I post them here?

What are my options here?

Thanks in advance for your help!

Martin-Laclaustra, 2020-11-02:
My first advice is to read the wiki:
https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_1
https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_2
Once you understand where your codes start and end, their bit-length and the length of the levels you can decide whether they are suitable for this library or not.
In the first case, at this point you should be able to create a protocol and use it. Example:
```
static const RCSwitch::Protocol customprotocol = { 810, { 4, 4 }, { 2, 2 }, { 1, 1 }, false };
mySwitch.setProtocol(customprotocol);
```
If it is not suitable you should look for help elsewhere.

Martin-Laclaustra, 2020-11-02:
I you just want to send your codes, have a look here:
https://github.com/sui77/rc-switch/issues/237#issuecomment-405937627

delphir-com, 2020-11-23:
Hi @Martin-Laclaustra,

am I understanding correctly, that this long-high in the beginning of the sequence is a "second" sync-bit which is not supported by the library?
also what is the technical issue for the support of the remotes with multiple sync-bits? 

[image]

```
808,500,376,948,380,948,384,940,820,484,824,484,824,484,820,488,388,940,388,940,392,932,828,480,396,932,25652,2808,796,512,796,512,792,516,792,512,796,512,364,960,796,512,796,512,796,508,796,512,796,512,796,512,796,508,796,512,796,512,364,960,800,508,800,508,368,956,804,504,800,508,368,960,372,952,804,504,376,948,808,500,376,948,380,948,384,940,816,492,816,492,816,492,812,496,812,496,812,496,808,500,380,944,384,940,820,488,816,492,384,944,384,944,388,936,824,484,820,488,820,488,820,488,388,936,392,936,392,936,824,484,392,936,25648,2828,768,540,768,544,764,540,768,540,772,524,352,972,788,520,788,520,784,524,784,520,788,520,784,524,780,532,776,528,780,524,352,976,784,524,784,520,356,972,788,512,796,516,360,964,364,964,796,512,364,960,800,508,368,952,380,944,384,940,824,484,820,488,820,488,820,488,816,488,820,488,816,492,384,944,388,936,824,484,820,488,388,936,396,932,396,932,828,480,824,484,824,484,824,484,388,936,396,932,396,928,832,480,396,928,25656,2824,772,540,768,540,768,536,772,536,772,532,344,980,780,528,780,528,776,540,764,544,764,536,776,524,784,524,780,528,780,524,356,968,788,520,788,516,360,964,796,516,788,528,344,980,348,980,784,520,352,976,784,524,352,972,360,968,360,960,800,512,796,512,792,516,792,512,792,516,792,516,796,504,368,964,364,960,804,500,804,504,372,952,380,952,372,952,812,492,816,492,812,496,812,496,380,944,384,944,384,940,824,488,388,936,25640,2844,756,552,756,552,756,548,760,548,756,552,324,1000,760,544,764,544,764,544,764,540,764,544,764,544,764,540,768,540,764,540,336,988,772,536,772,536,340,984,776,532,776,532,344,980,348,976,784,524,352,972,788,520,356,968,360,968,360,968,792,516,792,512,796,512,792,516,792,516,792,516,792,512,364,964,360,964,800,508,800,508,364,960,372,956,372,956,804,504,804,504,800,508,800,504,372,956,372,952,380,948,812,496,376,952,25640,2828,772,536,772,536,772,532,772,536,772,536,344,980,776,532,776,528,780,528,776,528,780,528,780,524,784,524,780,528,780,528,348,976,784,524,784,524,352,972,788,520,788,520,356,968,360,964,796,512,
```

delphir-com, 2020-11-23:
is there any technical issue to add the support of multiple sync-bits instead of a single sync-bit in the packet?

Martin-Laclaustra, 2020-11-23:
> is there any technical issue to add the support of multiple sync-bits instead of a single sync-bit in the packet?

Yes. Particularly for receiving codes. And structs are shared with emitting codes. Basically you need a new design to deal with it.

delphir-com, 2020-11-24:
is there any way/modification to make keeloq detectable? at least to show any output (even binary is fine) instead of being completely silent?

valentapat, 2022-03-12:
help me please, I'm trying add protocol into rcswitch.cpp (it is pool thermometer) Steinbach

I have signal:
8589,503,3822,501,1937,501,1892,501,3881,509,1915,509,1885,499,1915,501,1959,478,3824,485,1907,505,1912,504,3908,521,1895,502,1913,498,1891,502,1958,502,1936,501,1916,499,1940,497,1987,498,3824,498,3799,501,3822,479,1984,498,3847,478,3822,479,3823,500,1960,501,1938,499,1915,500,1937,499,1984,499,1918,498,1894,498,1917,497,1962,491,998,587,8610,504,3818,482,1912,503,1912,503,3910,481,1912,505,1933,503,1912,502,2003,503,3842,480,1912,502,1914,502,3910,526,1889,503,1913,502,1890,502,1980,502,1914,500,1916,499,1892,500,1983,499,3845,500,3844,480,3821,502,1959,501,3843,503,3820,502,3843,480,1980,504,1934,502,1891,502,1913,479,1981,501,1937,500,1892,501,1914,479,1981,492,1124,589,8586,505,3818,482,1911,504,1912,504,3908,482,1911,503,1935,503,1912,502,2002,503,3841,481,1912,503,1912,503,3909,481,1912,503

Try add code:
older
 { 500, {  1, 17 }, {  1,  4 }, {  1,  8 }, true },    // protocol 1

newest
{ 500,  0, { 0, 0 }, 1, { 1, 17 }, { 1, 4 }, { 1, 8}, true, 0  }

CSWITCH_SEPARATION_LIMIT. default, 10000 no efect
#define RCSWITCH_MAX_CHANGES 83 

it doesn't work, what is bad ???