sui77/rc-switch #136: CAME protocol

pmaster555, 2017-04-15:
How can I read and send data for the garage gate using this library?
[image]

In the screenshot the code is 3557

Martin-Laclaustra, 2017-04-15:
Check issue https://github.com/sui77/rc-switch/issues/103
That works with Raspberry pi and ESP8266 (it might fail with Arduino). Press your remote buttons long enough (3-5 seconds each).
If you want to measure manually you need to measure in milliseconds in your plot:
the width of the thin columns
the width of the thick columns (apparently double than the thin)
the width of the big silent gap.

then you write your protocol in the code:
{ thin width, { big gap/thin width, 1 }, { 1, 2 }, { 2, 1 }, true }

Your code is:
000110010010110111100101
or
111001101101001000011010
(I am not very familiar with inverted protocols)

If my universal receiver does not provide a suggestion, you already now how to view and measure the signal, so record and compare what you produce with what you want to get.
Please report your advances, successes and failures.
Best luck.

pmaster555, 2017-04-15:
The function of sending code looks like this ...

`void SendCame4(int Code) {
  for (int j = 0; j < 7; j++) {   // sending 7 times
    
    digitalWrite(2, HIGH);
    delayMicroseconds(320);
    digitalWrite(2, LOW);
    for (int i = 12; i > 0; i--) {
      byte b = bitRead(Code, i - 1);
      if (b) {
        digitalWrite(2, LOW); // 1
        delayMicroseconds(640);
        digitalWrite(2, HIGH);
        delayMicroseconds(320);
      }
      else {
        digitalWrite(2, LOW); // 0
        delayMicroseconds(320);
        digitalWrite(2, HIGH);
        delayMicroseconds(640);
      }
    }
    digitalWrite(2, LOW);
    delayMicroseconds(11520);
  }
}`

I added a string to RCSwitch.cpp

{ 320, { 36,  1 }, {  1,  2 }, {  2,  1 }, true },      // protocol 7 (Came)

But no reaction unfortunately does not happen ...

Martin-Laclaustra, 2017-04-15:
Your should allow the library to deal the timings for you instead of hard-coding them in your function.
try sending codes:
1650149
or
15127066
(check if I converted the binary codes correctly)
with the functions:
setProtocol(7)
and
send(1650149, 24)
or
send(15127066, 24)

I suggest that your record the signals after these two commands, and compare those signals to your original one. That can lead to the correct path.

When you succeed, that you will, please describe in detail your brand and model. Maybe even with photographs or links to seller.
It would be great if your could open your original remote and report which chipset it uses.

pmaster555, 2017-04-15:
This is the protocol of firm CAME
code on screen  =  3557
http://phreakerclub.com/447
Unfortunately there is nowhere to send. I can only read.

Martin-Laclaustra, 2017-04-16:
> How can I read and send data for the garage gate using this library?

So. Did you successfully detect your signal? (using the [examples](https://github.com/sui77/rc-switch/tree/master/examples))
Did you generate and send data successfully? (with the send function - also in the examples)
If not, what are you trying to achieve?

Martin-Laclaustra, 2017-04-16:
Please note that the right part of the code:
000110010010-110111100101 (1650149)
is exactly 3557

pmaster555, 2017-04-16:
The fact is that the library does not see this signal at all, as if it does not exist. No reaction to the buttons there ...
The standard on PT2262 sees normally...

pmaster555, 2017-04-16:
GENERAL
=======
Received 1650149
Bitlength 24
Protocol 0
Delay(short pulse length in us) 328
(delay from rc-switch is calculated from the sync gap and requires previous knowledge of the protocol)
(The following assumes protocols that detect the low period of sync)
(The next information is probably wrong for inverted protocols)
Sync bit high-low durations (in us) 286 21983

RAW SIGNAL (us)
===============
21983
310 321 569 323 562 332 563 575 330 555 357 320 551 345 555 568
383 296 567 331 560 558 342 328 540 620 313 582 289 365 549 572
316 565 335 603 296 587 306 349 546 360 540 608 286 360 532 602
286

STATISTICS
==========
Data bit duration = 897 and standard deviation (should be less than 10%) = 186.24
Do not use the rest of the information if big standard deviation
Long-to-short duration ratio for data bits (rounded) 2
Short duration (delay) recalculated from rounded ratio and average bit duration 299
Sync bit (in multiples of the previous short duration) 1 74

Proposed protocol for RCswitch
{ 299, { 1, 74 }, { 1, 2 }, { 2, 1 }, false }
Note: inverted protocol is not implemented

STATISTICS OF VARIATION BY LEVELS
=================================
(These are probably artifacts from detection delays or signal creation)
(might be completely ignored, but pay attention to them if big differences are present and emission does not work)
number of bits set (in 1650149): 12
longup, longdown, shortup, shortdown
369, 652, 505, 266
this might be useful for tweaking emmitting algorithms

=============================================

pmaster555, 2017-04-16:
RPi takes the code normally.
rc-swich on arduino Does not take :(

pmaster555, 2017-04-16:
The code turns out 24 bits and is equal all the same 1650149
But why can not he read the rc-swich???

Martin-Laclaustra, 2017-04-16:
try this protocol:
{ 299, { 74, 1 }, { 1, 2 }, { 2, 1 }, true }
this should work with sui77's rc-switch in Arduino for both, sending and receiving codes.

I have found that [my branch](https://github.com/Martin-Laclaustra/rc-switch/tree/protocollessreceiver) may not work in Arduino, although it does in RPi and ESP (thank you for testing and demonstrating that it works in RPi). I need to improve the output (it is in my roadmap).
I suspect that the problem is with variable lengths. May I ask you to test this new code (for my branch) in Arduino (I do not have one and I cannot test).
[RCSwitch.cpp.zip](https://github.com/sui77/rc-switch/files/924284/RCSwitch.cpp.zip)
There, all ints were substituted for longs in the function:
bool RECEIVE_ATTR RCSwitch::receiveProtocol(const int p, unsigned int changeCount)
but I think not all ints need to be replaced. May be, if this code works for you you could try int by int to see which one was causing problems and thus helping me in debugging the code. Thanks in advance.

pmaster555, 2017-04-16:
Probably I have a little confused your repositories and https://github.com/Martin-Laclaustra/rc-switch :)
Downloaded from yours and compiled for arduino issued an error

\libraries\rc-switch\RCSwitch.cpp: In static member function 'static bool RCSwitch::receiveProtocol(int, unsigned int)':

\libraries\rc-switch\RCSwitch.cpp:738:34: error: invalid initialization of reference of type 'RCSwitch::Protocol&' from expression of type 'const RCSwitch::Protocol'

         Protocol &pro = proto[i-1];

                                  ^

Martin-Laclaustra, 2017-04-16:
line 738 should read:
const Protocol &pro = proto[i-1];
I just patched it a couple of days ago.
but that line only applies to ESP8266. It should not even be read when compiling for Arduino.

pmaster555, 2017-04-16:
I'm doing on ESP8266

Martin-Laclaustra, 2017-04-16:
Then my code, the one in the repository, not the one in the zip, definitely works.
Anyway, try sui77's library with the protocol I wrote 3 comments ago.

pmaster555, 2017-04-16:
When compiled on arduino, ESB8266 gives an error
\libraries\rc-switch\RCSwitch.cpp: In static member function static bool RCSwitch::receiveProtocol(int, unsigned int):

\libraries\rc-switch\RCSwitch.cpp:738:34: error: invalid initialization of reference of type RCSwitch::Protocol& from expression of type const RCSwitch::Protocol

     Protocol &pro = proto[i-1];

Martin-Laclaustra, 2017-04-16:
Change this line:
line 738 should read:
const Protocol &pro = proto[i-1];

pmaster555, 2017-04-17:
Yes! Everything works fine, thanks for the help!

pmaster555, 2017-04-17:
Added a line to the library
{ 300, { 1,  73 }, {  1,  2 }, {  2,  1 }, false }      // protocol 7 CAME
But it is still defined as protocol 0
What have I done or have forgotten?

Martin-Laclaustra, 2017-04-18:
I assume that you are using my fork. The recognition loosely tries to match with a protocol. That algorithm is not exact. In fact the fork is about recognizing without protocol.
However, to send codes, you will need to use protocol 7.
If you use sui77's branch... does it recognize your signals as protocol 7?

pmaster555, 2017-04-18:
Does not recognize the protocol :(

Martin-Laclaustra, 2017-04-19:
Does SENDING with that protocol 7 get recognized by your socket? (you can test with any branch, as the sending part of the library is identical for both)
A) If so, then may be your remote emits with quite a lot of irregularity in the lengths (long ups 530-570, long downs 550-620, short ups 285-385, short downs 295-365) and sui77's library is quite stringent with respect to that. Just use my branch for receiving and check for your code. Ignore the protocol when receiving.
B) If not recognized by your socket, please record the signal sent by rc-switch with that protocol 7, like in your initial post, to compare the original remote signal and this "synthetic" one, to search for differences.

giangel84, 2017-06-03:
I'm trying to set a new protocol into RCswitch.cpp in order to receive 13 bit code.

The type of code is 13 bit ( bit start = 0 , 10 bit of  code , bit stop = 0 , bit stop =1). 
BIT 0 = 300µs , BIT 1 = 600µs.
example : 0 1000010101 01

How should be the protocol? Is it possible to make one?
Thank you

Martin-Laclaustra, 2017-06-06:
@giangel84 rc-switch does not look like the library you need (or you did not record/describe the signal adequately). This question does not belong here. It is a different issue. Show a recorded signal in a new issue. Have you tried the above links to automatically recognize the signal?

giangel84, 2017-06-07:
Yes I tried but I managed to receive only codes from other RC remote control, not mine that is a TOP434EV (CAME 4 channel remote).
I will get an Logic signal analyzer to understand my signal's logic and will try again.

Martin-Laclaustra, 2017-06-07:
use https://github.com/sui77/SimpleRcScanner

CarlosLB, 2017-07-04:
I have a CAME board with RF receiver and TAM432SA remotes. I'll try to use the information here to decode it. I have Arduino and ESP8266 (Node MCU) boards to try.

pmaster555, 2017-07-05:
http://skproj.ru/otkryt-shlagbaum-came/

CarlosLB, 2017-07-13:
I've used Martin's library to receive and this is what I got.
Proposed protocols only work in very short range (antenna included).
Any idea?

Received 12490478 / 24bit Protocol: 0
data bits of pulse train duration: 22604
proposed protocol: { 314, { 73, 3 }, { 1, 2 }, { 2, 1 }, true }
====
first level down
22824 824 
508 492 440 444 488 376 556 360 568 356 580 384 548 424 236 744 
508 460 200 816 164 696 544 368 300 684 568 376 552 420 236 724 
532 428 500 396 536 440 212 684 572 388 536 340 600 336 316 668 

====
Received 16684782 / 24bit Protocol: 0
data bits of pulse train duration: 22836
proposed protocol: { 317, { 72, 3 }, { 1, 2 }, { 2, 1 }, true }
====
first level down
22820 824 
560 404 552 408 544 396 560 368 588 380 576 392 548 432 244 700 
532 476 192 768 192 716 516 408 256 672 564 400 556 412 248 684 
568 380 556 392 564 372 308 676 560 364 588 376 564 384 296 644 

====
Received 16684782 / 24bit Protocol: 0
data bits of pulse train duration: 22660
proposed protocol: { 315, { 73, 3 }, { 1, 2 }, { 2, 1 }, true }
====
first level down
22868 824 
616 328 596 324 608 320 616 312 616 312 620 320 612 336 328 652 
600 312 348 724 252 632 620 324 336 672 572 336 596 324 328 636 
620 312 620 312 620 316 344 624 624 308 624 300 628 304 360 616 

====
Received 16684782 / 24bit Protocol: 0
data bits of pulse train duration: 22648
proposed protocol: { 315, { 72, 3 }, { 1, 2 }, { 2, 1 }, true }
====
first level down
22836 828 
564 400 528 416 520 344 580 352 580 364 572 348 584 376 284 728 
520 372 288 700 280 656 588 404 260 728 520 420 512 368 288 724 
528 364 568 352 576 380 284 696 548 368 568 344 584 332 332 656 

====
Received 16684782 / 24bit Protocol: 0
data bits of pulse train duration: 22596
proposed protocol: { 314, { 73, 3 }, { 1, 2 }, { 2, 1 }, true }
====
first level down
22828 824 
500 376 556 364 564 420 512 348 588 364 564 348 580 376 284 640 
608 380 284 644 332 680 576 332 324 652 600 352 576 388 272 728 
516 388 548 428 504 356 304 700 548 380 552 348 584 376 284 668 

====

Martin-Laclaustra, 2017-07-15:
@CarlosLB 
use https://github.com/sui77/SimpleRcScanner
record the signals that you generate and those from the original transmitter: compare those signals. You will see whether there are timing differences.
If they aren't. You could try to send more repeats (default is 4). Check that your antenna is in the correct pin.

DarkDaemon1, 2017-07-21:
I have a 12-bit CAME transmitter and well working SENDING code:

```
#define txPin 3
#define Te 320

void setup(){
  Serial.begin(9600);
  Serial.println("Came send...");
  pinMode(txPin, OUTPUT);
  SendCame(3333); //for example code is 3333 (variants are from 0 to 4095)
  Serial.println("success");
}

void loop(){
  
}

void SendCame(unsigned long code){     
  for (int j=0;j<10;j++) {
    //Start impulse
    digitalWrite(txPin,LOW);
    delayMicroseconds(Te*36);
    digitalWrite(txPin,HIGH);
    delayMicroseconds(Te);
    //Sending every bit
    for (byte i=0;i<12;i++){
      byte b = bitRead(code, i);
      digitalWrite(txPin,LOW);
      delayMicroseconds(Te);
      if (!b) digitalWrite(txPin,HIGH);
      delayMicroseconds(Te);
      digitalWrite(txPin,HIGH);
      delayMicroseconds(Te);
    }
    digitalWrite(txPin,LOW);
    delay(16);
  }
}
```
And I want to add support of CAME to RCSwitch library. I added to RCSwitch.cpp this code:
```
  { 320, { 36,  1 }, {  1,  2 }, {  2,  1 }, true },     // protocol 7 (Came) Holtek HT-12E
  { 700, { 36,  1 }, {  1,  2 }, {  2,  1 }, true }      // protocol 8 (Nice) Holtek HT-12E
```
but transmitter don't receive the code :(
`mySwitch.send(3333, 12);`
 
Could you please explain me, why in "Send" procedure first is the sendind code, and the second is the sending Sync bit? I think this shoud be contrariwise.

superyarik, 2017-09-14:
@Martin-Laclaustra hi, i've check working of your protocol on your branch, it still don't read CAME signal.

i have code for reading (and send) without libraries, could you pleace view and mayme be edit protocol 7 for CAME?

https://gist.github.com/superyarik/3eb4da9da728466c072e716532d732ef

aitia, 2017-12-18:
any update please? I can not catch came remote with rc-switch simple/advanced receive

fugaldo, 2018-11-06:
Hello @DarkDaemon1 
  the remote TOP432M is recognized...
Received xxxx / 12bit Protocol: 6
data bits of pulse train duration: 12181
proposed protocol: { 339, { 34, 1 }, { 1, 2 }, { 2, 1 }, true }

Nothing for the remotes TOP432NA/TOP432EE and above.

dauheeIRL, 2020-04-27:
Hi, I can confirm that the addition of { 320, { 36,  1 }, {  1,  2 }, {  2,  1 }, true } protocol allows me to control my CAME gate using TOP432M. Could this and the other protocols above be added to the library?

I attempted to perform this action on my CAME gate 5 years ago and didn't have the skills and/or commitment to decode the signal. Am so happy someone has figured this out. Fantastic work

1technophile, 2020-04-27:
Hi, yes you can submit a PR with them if you are at least 2 confirming they are working.

Benas09, 2022-02-10:
I have successfully cloned TOP432EV code using 433 receiver and sound card.

Here is the code https://github.com/Benas09/came/blob/master/remote.ino

As far I have only one distinct code, i can not make function universal. 

I send code 1011101101011**011010100011**
Firmware from http://skproj.ru/otkryt-shlagbaum-came/ shows that my code is 1699 (**011010100011** in binary). 
So I dont know if first part of code is unique or not.

ashumkin, 2023-05-10:
I have two TOP432NA remotes which are read with RX-470C-V01 (WL-101) receiver and the code from http://skproj.ru/otkryt-shlagbaum-came/  read left buttons as 1039 (yes, both) and right buttons as 1115 and 281, respectively.
I've managed to emit successfully the same codes with my transmitter (WL-102) (with the sketch from the URL above) and a barrier has accepted them (hurah!).
Then I've decided to use rc-switch library only (I want to send two different codes (for different gates) with the same single codebase) and I've tried to reproduce the same signals with it.
I have got the success with the 'builtin' protocol 11 (as a my receiver _sometimes_ decodes signal emitted with my transmitter (not a remote) with that protocol ID) - the barrier accepts signal sent but my receiver does not decode the original TOP432NA remote's signal with that protocol (nor any protocol).

So, I've tried @Martin-Laclaustra's (big thanks!) [ProtocolAnalyzeDemo.ino](https://github.com/Martin-Laclaustra/rc-switch/blob/protocollessreceiver/examples/ProtocolAnalyzeDemo/ProtocolAnalyzeDemo.ino), it has suggested
 `{ 323, { 45, 2 }, { 1, 2 }, { 2, 1 }, true }` parameters, and voi-la! Now, my receiver decodes both the original's remote and my transmitter's signals!