sui77/rc-switch #138: Compilation failes on Raspberry PI

SamDecrock, 2017-04-29:
Hello,

From the code and commit history, I see that there should be support for Raspberry Pi.

However when compiling, I'm getting:

```
/tmp/ccKsGhAS.o: In function `RCSwitch::enableTransmit(int)':
RCSwitch.cpp:(.text+0x214): undefined reference to `pinMode'
/tmp/ccKsGhAS.o: In function `RCSwitch::transmit(RCSwitch::HighLow)':
RCSwitch.cpp:(.text+0x1020): undefined reference to `digitalWrite'
RCSwitch.cpp:(.text+0x1038): undefined reference to `delayMicroseconds'
RCSwitch.cpp:(.text+0x1050): undefined reference to `digitalWrite'
RCSwitch.cpp:(.text+0x1068): undefined reference to `delayMicroseconds'
/tmp/ccKsGhAS.o: In function `RCSwitch::enableReceive()':
RCSwitch.cpp:(.text+0x10f0): undefined reference to `wiringPiISR'
/tmp/ccKsGhAS.o: In function `RCSwitch::handleInterrupt()':
RCSwitch.cpp:(.text+0x1570): undefined reference to `micros'
collect2: error: ld returned 1 exit status
```

Even though I added `#define RPI 1` at the beginning of RCSwitch.h

These are all Arduino related functions (`digitalWrite`, `delayMicroseconds`, ...) Could it be that I'm missing something on my Pi to emulate these functions?

Sam

SamDecrock, 2017-04-29:
Nevermind, I adapted the Makefile from https://github.com/ninjablocks/433Utils and now it works.
(My c++ skills are still in development ;-) )

closed 2017-04-29 by SamDecrock

DanChaltiel, 2018-08-20:
Hi @SamDecrock, could you share your Makefile ? I have been trying to understand this for hours now (never used c++ before). Thanks !

SamDecrock, 2018-08-22:
Well, this is my Makefile:

```Makefile

# Defines the RPI variable which is needed by rc-switch/RCSwitch.h
CXXFLAGS=-DRPI

all: senddemo receivedemo receivedemo_json

senddemo: rc-switch/RCSwitch.o senddemo.o
	$(CXX) $(CXXFLAGS) $(LDFLAGS) $+ -o $@ -lwiringPi

receivedemo: rc-switch/RCSwitch.o receivedemo.o
	$(CXX) $(CXXFLAGS) $(LDFLAGS) $+ -o $@ -lwiringPi

receivedemo_json: rc-switch/RCSwitch.o receivedemo_json.o
	$(CXX) $(CXXFLAGS) $(LDFLAGS) $+ -o $@ -lwiringPi

clean:
	$(RM) *.o senddemo receivedemo receivedemo_json

```

In this Makefile, I define 4 'commands': 3 of the 4 are for building 3 of my 3 demo's. The other one is for removing my 3 demo's. My 3 demos are 'senddemo.cpp', 'receivedemo.cpp' and 'receivedemo_json.cpp'

For example, receivedemo.cpp looks like this:

```cpp
#include <stdio.h>
#include "rc-switch/RCSwitch.h"

int main(int argc, char *argv[]) {

    printf("Starting receivedemo\n");
    printf("Make sure you have connected the 433Mhz receiver data pin to WiringPi pin 2 (real pin 13). See: https://pinout.xyz/pinout/pin13_gpio27\n");
    printf("\n");
    
    int PIN = 2; // this is pin 13, aka GPIO22 on the PI3, see https://www.element14.com/community/servlet/JiveServlet/previewBody/73950-102-10-339300/pi3_gpio.png

    if (wiringPiSetup () == -1) {
        printf("ERROR: WiringPi not installed. Make sure you have WiringPi installed.\n");
        printf("Quick tutorial:\n\n");
        printf("    sudo apt-get install git\n");
        printf("    cd ~/\n");
        printf("    git clone git://git.drogon.net/wiringPi\n");
        printf("    cd wiringPi\n");
        printf("    ./build\n\n");
        return 1;
    }

    // put the PIN into no-pull/up down state:
    // see https://github.com/ninjablocks/433Utils/issues/21
    pullUpDnControl(PIN, PUD_OFF);

    RCSwitch mySwitch = RCSwitch();

    mySwitch.enableReceive(PIN);

    printf("Listening\n");

    while(true) {

        if (mySwitch.available()) {
            
            int value = mySwitch.getReceivedValue();
            
            if (value == 0) {
                printf("Unknown encoding\n");
            } else {
                printf("Received %lu / %i bit Protocol: %i\n", mySwitch.getReceivedValue(), mySwitch.getReceivedBitlength(), mySwitch.getReceivedProtocol());
            }

            mySwitch.resetAvailable();
        }
    }
}
```

Note that the rc-switch source files are in a subfolder called `rc-switch` 

To compile, just run `make`. To clean, run `make clean`. This seems to be a good introduction into makefiles: https://www.youtube.com/watch?v=_r7i5X0rXJk