sui77/rc-switch #141: How does your library distinguish between a low signal and a high signal?

SamDecrock, 2017-05-05:
Hello,

Thanks for your cool library, it really helps me in learning these 433Mhz protocols.

I began reading your library and I can't seem to figure out how you distinguish between an incoming low signal and an incoming high signal.

As I understand it `handleInterrupt()` gets called when the signal changes from low to high or from high to low. But I never see you querying the pin. So how can you know a message started with a high or a low signal? Or do you just assume the signal is always low until a new message comes in?

Thanks!
Sam

Martin-Laclaustra, 2017-05-05:
Currently, master branch is not aware of that.
You can check the [protocollessreceiver branch](https://github.com/Martin-Laclaustra/rc-switch/tree/protocollessreceiver) for a code that stores the level of the first period to enable further classification of the incoming protocols.

SamDecrock, 2017-05-09:
Thanks! How does the protocol get the detected now? Does it just assume the first signal is high?

Martin-Laclaustra, 2017-05-18:
It pays attention only to the timings disregarding whether they are high or low.
For example, these two signals get recognized as the code 1234 (protocol 1, 24 bits) (tested):
[image]
[image]
The first signal is the first long timing above a duration threshold that appears twice within a certain timeframe. Then it is assumed high or low (i.e. is ignored) depending which protocol is the one that matches the timings between these long signals.

SamDecrock, 2017-05-19:
All right.

Thanks for this insightful answer!

closed 2020-01-05 by 1technophile