sui77/rc-switch #142: Problem receiving rc si4010c2

pablo-pasion, 2017-05-07:
Hi,

I need to decode RF 433mhz signals for a set of remote control.
As far as I see,  Remote control is a cheap model 4 button and have a si4010c2 chipset. (not supported)

I have the following

- Arduino UNO
- 433Mhz receiver

I have connected

- Receiver data pin to  arduino digital 2
- gnd to gnd
- vcc to 5v

I'm using rc-switch library demo-receive-simple and advanced sketch without success.
I'cant see anything throught serial port 9600.

Any Ideas? 
What Can I do?

Thanks in advance
Regards

Pablo

[image]

Martin-Laclaustra, 2017-05-08:
Check the first answer to issue (https://github.com/sui77/rc-switch/issues/135).
1. Test if the receiver is correctly outputing values to your pin (with the simple scanner).
2. If it is and your protocol is compatible with rc-switch, the _protocollessreceiverbranch_ will suggest a protocol to test.

xnaron, 2018-02-06:
Did you ever get this working?  If so am interested in some details as the Lutron Maestro remotes use this same chip.  Thanks!

jrmusan, 2020-12-07:
Did you ever figure this out? There's gotta be a way to get this chip working, I've found some code that recognizes the signal but isn't decoding it correctly.

Martin-Laclaustra, 2020-12-13:
@jrmusan please, if you want to receive help with this, read in the wiki the two chapters on "how to add a new remote" and then post as much relevant information as possible in this thread.