sui77/rc-switch #143: Receive and ATiny85

danjohn, 2017-05-10:
The implementation disables receive features on ATtiny's due to "missing libm depencies (udivmodhi4)".

I don't know, may be that requirement has changed in the last few months (new libs? adapted code?), but I can definitely compile and (obviousely successful) run that receive code on a DigiSpark-like target.

Hopefully this could be verified by a more experienced user!?
(Also, I can't see where in the sources "divmod" would be used (pls give a noob a hint!:))

All I did, was to comment out "#define RCSwitchDisableReceiving" in my local RCSwitch.h line #56.
Sketch:
'''#include <RCSwitch.h>
RCSwitch mySwitch = RCSwitch();
const int LED = 1;

void setup() {                
  mySwitch.enableReceive(0);  // Receiver on interrupt 0 => that is pin #2
  pinMode(LED,OUTPUT);
}

void loop() {
   if (mySwitch.available()) {
      digitalWrite(LED,HIGH);
      mySwitch.resetAvailable();
   } else {
      digitalWrite(LED,LOW);
   }
   delay(10);
}
'''

IDE: arduino-1.8.2-linux32
Target: <http://www.electrodragon.com/product/attiny85-usb-mini-dev-board-digispark-compatible/>

rolandas-valantinas, 2018-03-01:
Which pin on Attiny85 you use for receiving?
Though code does compile with https://github.com/damellis/attiny LED is not blinking while I'm transmitting from Arduino.

bodtx, 2018-03-19:
According to his comment

>  // Receiver on interrupt 0 => that is pin #2

And to the[ pin definition ](https://mwwalk.files.wordpress.com/2013/05/attiny-pinout.png)

The pin to use is 2, the one under VCC.

I'll check that tonight as I would like to use attiny as receivers also. Will give a feedback

bodtx, 2018-03-19:
Ok, I've just did a test like this and led is blinking on attiny85 as receiver:

```
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {

  // Transmitter is connected to Arduino Pin #10  
  mySwitch.enableTransmit(10);

}

void loop() {
  mySwitch.send(5393, 24);
  delay(1000);  
  mySwitch.send(5396, 24);
  delay(1000);  

  
}
```

lindenaar, 2019-05-01:
I can confirm that there seems to be no problem to simply comment out the "#define RCSwitchDisableReceiving" in RCSwitch.h (line 56). Using interrupt 0 (pin 2) is working on Digispark with ATTiny85 just like on an Arduino Nano or Leonardo. 

I will continue testing while working on a project that sends and receives (not to itself) over RF433 and so far that is working fine on this platform (I have not encountered any issues yet) so suggest to take out this #define at least for the __AVR_ATtinyX5__ platform.

quanticchaos, 2019-10-11:
I confirm that commenting the above said line also works with bare attiny85. But not with every library. Didn't work with Spence Konde "[ATTinyCore](http://drazzy.com/package_drazzy.com_index.json)" Library but did work when using "[attiny](https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json)" Library by David A. Mellis

coldsilk, 2021-04-17:
I'm using very old attiny45's and 85's from their release 16 years ago and using David A. Mellis's cores the 45's work, but the 85's do not. Neither work with Spence Konde's cores. These old 45's just work straight out of the tube without burning a bootloader or anything, but these 85's won't work no matter what I try.

bodtx, 2021-04-17:
Did yout try to change the speed, 1Mhz, 8Mhz? Do they fail while uploading or when running?

coldsilk, 2021-04-19:
Yes, and I was WRONG. Both the 45's and 85's work at both 1 and 8 using Mellis's cores. So Mellis's seems choice.

I just received new 85's but I don't know what I changed in the last few days because now when I try to burn a bootloader on the new or old 85's it states: "free(): invalid next size (normal)". Again though, Mellis's works just fine.

vi7, 2021-05-18:
I confirm receiving is working on the ATTiny85 with the following setup:
* board:
  * type: Digispark USB clone (rev.3 with LED on PB1)
  * bootloader: [Micronucleus v2.5](https://github.com/micronucleus/micronucleus/blob/v2.5/firmware/releases/t85_default.hex)
  * CPU frequency: 16.5MHz (default for this board, calibrated by the Micronucleus bootloader from USB)
* IDE and libs:
  * VSCode + PlatformIO 5.1.1
  * core: [DigistumpArduino](https://github.com/ArminJo/DigistumpArduino) 1.7.2 - configured as default by the PlatformIO when choosing platform `atmelavr` and board `digispark-tiny`
  * RCSwitch v2.6.4 with commented `RCSwitchDisableReceiving`

So as a conclusion Mellis core is definitely not the only one option

@sui77 assuming that multiple people confirm ATTiny85 receive is working pls consider removing `RCSwitchDisableReceiving` at least for ATTinyX5, I can create PR for that if you'd like or otherwise users of your awesome lib will have to use custom fixed forks all over the place 😢

jilvir, 2023-09-13:
Hello, I too late here... I just restarted now hehe... I need same, I use digistump Attiny85. Blue with the micro USB. 
When I used the Arduino Uno it is fine I have no issue, but my focus is on ATtiny85 because that is more than enough hardware for my project and I love it.

First issue that I encountered is compiling with the RCSwitch Library and saw online regarding the solution same above that I commented the line #define RCSwitchDisableReceiving... 
I used the Digistump | DigiSpak board in Arduino IDE compiling and uploading Successful but when I pressed the remote key (433 MHZ) there is no response of the built-in LED which I call for the high and low of the LED to know if there is a signal detected. It means its not working. The INT0 in my board is PB2 pin# 7 in the actual IC. PB1 has the built-in LED.

So... I follow here... excitedly...

1. I install download the ATtiny Library based on the link above and install
2. Copy and paste the sketch of @bodtx
3. Select the board under attiny | ATtiny25/45/85
4. Selecting the clock 16MHz Internal
5. Processor: ATtiny85
6. Programmer (??? I don't know what to select, I didn't see the Micronucleus which I always used)
7. I compile the sketch and successful
8. but in uploading it gives me an error "A programmer is required to upload"

@vi7 and @bodtx can you help?

jilvir, 2023-09-13:
I try to this board 
https://www.aliexpress.com/item/33062782242.html?src=google&src=google&albch=shopping&acnt=897-548-8760&slnk=&plac=&mtctp=&albbt=Google_7_shopping&gclsrc=aw.ds&albagn=888888&ds_e_adid=&ds_e_matchtype=&ds_e_device=c&ds_e_network=x&ds_e_product_group_id=&ds_e_product_id=ar33062782242&ds_e_product_merchant_id=106258787&ds_e_product_country=SA&ds_e_product_language=ar&ds_e_product_channel=online&ds_e_product_store_id=&ds_url_v=2&albcp=20223361648&albag=&isSmbAutoCall=false&needSmbHouyi=false&gclid=CjwKCAjwu4WoBhBkEiwAojNdXkU_NzKthkJ8eN9H8Fo4KIyIJ6IAUhHpDm8YOu9FyuhbhEE9sC5KDxoCJG0QAvD_BwE&aff_fcid=9e294b7203a443259b3c6e5c32fadf5e-1694627064233-08135-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=9e294b7203a443259b3c6e5c32fadf5e-1694627064233-08135-UneMJZVf&terminal_id=41ac2a7ef8714a1a9ae9a126a1a04dac&afSmartRedirect=y

and same as yours @vi7 

https://www.aliexpress.com/item/33062782242.html?src=google&src=google&albch=shopping&acnt=897-548-8760&slnk=&plac=&mtctp=&albbt=Google_7_shopping&gclsrc=aw.ds&albagn=888888&ds_e_adid=&ds_e_matchtype=&ds_e_device=c&ds_e_network=x&ds_e_product_group_id=&ds_e_product_id=ar33062782242&ds_e_product_merchant_id=106258787&ds_e_product_country=SA&ds_e_product_language=ar&ds_e_product_channel=online&ds_e_product_store_id=&ds_url_v=2&albcp=20223361648&albag=&isSmbAutoCall=false&needSmbHouyi=false&gclid=CjwKCAjwu4WoBhBkEiwAojNdXkU_NzKthkJ8eN9H8Fo4KIyIJ6IAUhHpDm8YOu9FyuhbhEE9sC5KDxoCJG0QAvD_BwE&aff_fcid=9e294b7203a443259b3c6e5c32fadf5e-1694627064233-08135-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=9e294b7203a443259b3c6e5c32fadf5e-1694627064233-08135-UneMJZVf&terminal_id=41ac2a7ef8714a1a9ae9a126a1a04dac&afSmartRedirect=y

bodtx, 2023-09-13:
Never use this board but if as I understand you use Arduino IDE and you want to upload with a programmer, you have to choose it here
[image]

jilvir, 2023-09-14:
Yes... I'm using Arduino... So I will use the ATtiny85 without the board which I attached the link? I will use the ATtiny85 microcontroller IC only not with the board and upload the sketch using other Arduino board and upload it using Arduino as ISP?

because the ATtiny85 with the board as same as I attach the link I don't need Arduino Board it can be directly upload the sketch with the programmer Micronucleus

jilvir, 2023-09-15:
> Never use this board but if as I understand you use Arduino IDE and you want to upload with a programmer, you have to choose it here [image]

@bodtx I found out that the error message "A programmer is required to upload." is when I used the Upload button and the uploading successful when I'm using the Sketch | Upload Using Programmer.

so again I uploaded your code and nothing happened. I put the data pin of the RF receiver to the PB2 (pin 7 physical pin of the ATtiny85 next/under the VCC pin where the INT0 pin) and the LED on the PB1 (pin 6 physical pin on the ATtiny85).

Any help please? can you share your's how did you do it?

jilvir, 2023-09-15:
> Ok, I've just did a test like this and led is blinking on attiny85 as receiver:
> 
> ```
> #include <RCSwitch.h>
> 
> RCSwitch mySwitch = RCSwitch();
> 
> void setup() {
>   mySwitch.enableReceive(0);  // Receiver on interrupt 0 => that is pin #2
>   pinMode(4, OUTPUT);
> }
> 
> void loop() {
>   if (mySwitch.available()) {
>     int value = mySwitch.getReceivedValue();
>     if (value == 5393) {
>       digitalWrite(4, HIGH);
>     }
>     else if (value == 5396) {
>       digitalWrite(4, LOW);
>     }
>     mySwitch.resetAvailable();
>   }
> }
> ```

This is the code I uploaded, I want to see only if there is a reaction when pressing any RF remote or detects any signal. just to test if it is working the code and library.

```
#include <Arduino.h>
#include <RCSwitch.h>
#define RF_PIN PB2
#define LED PB1
RCSwitch mySwitch = RCSwitch();

void setup() 
{                
  mySwitch.enableReceive(RF_PIN);  // Receiver on interrupt 0 => that is pin #2
  pinMode(LED,OUTPUT);
}

void loop() {
   if (mySwitch.available())
  {
    digitalWrite(LED, HIGH);
    mySwitch.resetAvailable();
    delay(100);
   } else 
   {
    digitalWrite(LED, LOW);
   }
} ```

bodtx, 2023-09-15:
Did it work on a real Arduino, uno for exemple ?

jilvir, 2023-09-15:
> Did it work on a real Arduino, uno for exemple ?

No... I try in Arduino Uno but still not working....
So... what I did is uncommented the #if defined( __AVR_ATtinyX5__ ) or defined ( __AVR_ATtinyX4__ ) and #endif and live the 
//#define RCSwitchDisableReceiving commented and try again with Arduino Uno and it works... 

In my ATtiny85, I change the #define RF_PIN PB2 to RF_PIN 0 and upload the sketch and yehhheyyy it working...

So... I found out that mySwitch.enableReceive(0) must always be zero parameter but you will connect the RF data pin the where is the INT0 in your kind of board and I uncomment also the #if defined( __AVR_ATtinyX5__ ) or defined ( __AVR_ATtinyX4__ ) and #endif and live the  //#define RCSwitchDisableReceiving commented.

Now, it's perfectly working... and my project was done!

Thank you soo much @bodtx I can't solve it with out you... I really appreciated your help...

bodtx, 2023-09-15:
Very good, so you were missing some defined parameters. 
Happy it  helped, but as always you did most of the job 🤯🤩😎

jilvir, 2023-09-15:
> Very good, so you were missing some defined parameters. Happy it helped, but as always you did most of the job 🤯🤩😎

yes... and as always... thank you soo much...

vi7, 2023-09-15:
hey @jilvir , as I see I'm a bit late for the party. JFYI feel free to check [my sample project](https://github.com/vi7/arduino-rx-test) for RX testing on different boards including our subj - Digispark ATTiny85 USB. Couple of notes about the project:
- it's [PlatformIO](https://platformio.org) based - I really encourage using it instead of Arduino IDE
- it uses my fork of RCSwitch which already has needed code changes for the ATTiny85
- has examples of Serial logging on ATTiny on various cores (which might be reeeally helpful sometimes when LED is just not enough)

Worth to mention that my ATTiny85 USB clone has `INT0` interrupt on `PB2` pin, that is the one used by default when you do `receiver.enableReceive(0)` . On the other hand project above allows to use you any ATTiny85 pin with some additional interrupt messing code (included in the example)

jilvir, 2023-09-15:
> ing code (included in the example)

Thank you soo much... yes I will... thanks...

nicholasareed, 2024-09-09:
For anyone checking, this also works with an ATtiny84 (just make sure to use the *ONLY* external interrupt pin for your DATA pin. It is physical pin 5, aka "INT0" aka "PB2"), and you need to comment out `#define RCSwitchDisableReceiving` on line 56 of RCSwitch.h, as mentioned.