sui77/rc-switch #145: Support for CC1101

martenumberto, 2017-05-12:
Is the CC1101 supported by this Library?
I have an Arduino Nano with a CC1101 Chip.
https://www.amazon.com/Solu-Wireless-Transceiver-Antenna-Arduino/dp/B00XDL9838/ref=sr_1_3?ie=UTF8&qid=1494574220&sr=8-3&keywords=cc1101&th=1

Locke, 2017-12-18:
The CC1101 is not supported by this library. It doesn't provide raw access to the sender/receiver as this library assumes, as it uses it's own package format and has some more complex internal logic to calculate things like a checksum, see http://www.ti.com/lit/ds/symlink/cc1101.pdf#page=37.

It might be worth looking at https://github.com/SpaceTeddy/CC1101. I haven't tried it out though..

LSatan, 2018-03-15:
Hello,
the theme is a bit older but
it works great with cc1101.
the cc1101 only needs to be configured correctly. got it done the rx demo of rc switch to get running. the nice thing about the cc1101 is that it not only works with 433 mhz. I'm just about to implement the rest. Maybe it helps others.

LSatan, 2018-03-16:
I managed to make the cc1101 compatible with rcswitch. I have made a kind of driver so that it works. since I have rewrite a lib for someone else I have to wait now for permission to publish it.
Sorry for my Bad english.

LSatan, 2018-03-16:
The Driver is ready and Works Perfect.

LSatan, 2018-03-18:
here is the driver:
https://github.com/LSatan/RCSwitch-CC1101-Driver-Lib

YindSoft, 2018-10-15:
> here is the driver:
> https://github.com/LSatan/RCSwitch-CC1101-Driver-Lib

LSatan, you're a saint!

1technophile, 2020-01-05:
Closing the issue as CC1101 is supported by @Lsatan library

closed 2020-01-05 by 1technophile